xoseperez/espurna #618: Sonoff Dual R2 error compiling

skyynet, 2018-03-03:
After successfully flashing some Sonoff Dual R1 I now bought an R2. When I try to flash it (after having changed the config in arduino.h to R2 of course) I get errors while compiling
`Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DIO, 1M (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

/Users/name/espurna/code/espurna/button.ino: In function 'void buttonSetup()':
button:137: error: no matching function for call to 'DebounceEvent::DebounceEvent(int, int, int, long unsigned int&)'
             _buttons.push_back({new DebounceEvent(BUTTON1_PIN, BUTTON1_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON1_RELAY});
                                                                                                             ^
/Users/name/espurna/code/espurna/button.ino:137:109: note: candidates are:
In file included from /Users/name/espurna/code/espurna/button.ino:13:0:
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:46:9: note: DebounceEvent::DebounceEvent(uint8_t, uint8_t, long unsigned int)
         DebounceEvent(uint8_t pin, uint8_t mode = BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:46:9: note:   candidate expects 3 arguments, 4 provided
In file included from /Users/name/espurna/code/espurna/button.ino:13:0:
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:45:9: note: DebounceEvent::DebounceEvent(uint8_t, DebounceEvent::TDebounceEventCallback, uint8_t, long unsigned int)
         DebounceEvent(uint8_t pin, TDebounceEventCallback callback, uint8_t mode = BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:45:9: note:   no known conversion for argument 2 from 'int' to 'DebounceEvent::TDebounceEventCallback {aka std::function<void(unsigned char, unsigned char, unsigned char, short unsigned int)>}'
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note: DebounceEvent::DebounceEvent(const DebounceEvent&)
 class DebounceEvent {
       ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note:   candidate expects 1 argument, 4 provided
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note: DebounceEvent::DebounceEvent(DebounceEvent&&)
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note:   candidate expects 1 argument, 4 provided
button:137: error: no matching function for call to 'std::vector<button_t>::push_back(<brace-enclosed initializer list>)'
             _buttons.push_back({new DebounceEvent(BUTTON1_PIN, BUTTON1_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON1_RELAY});
                                                                                                                                       ^
/Users/name/espurna/code/espurna/button.ino:137:135: note: candidates are:
In file included from /Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/vector:64:0,
                 from /Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/random.h:34,
                 from /Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:50,
                 from /Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from /Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from /Users/name/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/Arduino.h:240,
                 from sketch/espurna.ino.cpp:1:
/Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h:901:7: note: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = button_t; _Alloc = std::allocator<button_t>; std::vector<_Tp, _Alloc>::value_type = button_t]
       push_back(const value_type& __x)
       ^
/Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h:901:7: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const value_type& {aka const button_t&}'
/Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h:919:7: note: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = button_t; _Alloc = std::allocator<button_t>; std::vector<_Tp, _Alloc>::value_type = button_t]
       push_back(value_type&& __x)
       ^
/Users/name/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h:919:7: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'std::vector<button_t>::value_type&& {aka button_t&&}'
button:143: error: no matching function for call to 'DebounceEvent::DebounceEvent(int, int, int, long unsigned int&)'
             _buttons.push_back({new DebounceEvent(BUTTON2_PIN, BUTTON2_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON2_RELAY});
                                                                                                             ^
/Users/name/espurna/code/espurna/button.ino:143:109: note: candidates are:
In file included from /Users/name/espurna/code/espurna/button.ino:13:0:
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:46:9: note: DebounceEvent::DebounceEvent(uint8_t, uint8_t, long unsigned int)
         DebounceEvent(uint8_t pin, uint8_t mode = BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:46:9: note:   candidate expects 3 arguments, 4 provided
In file included from /Users/name/espurna/code/espurna/button.ino:13:0:
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:45:9: note: DebounceEvent::DebounceEvent(uint8_t, DebounceEvent::TDebounceEventCallback, uint8_t, long unsigned int)
         DebounceEvent(uint8_t pin, TDebounceEventCallback callback, uint8_t mode = BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:45:9: note:   no known conversion for argument 2 from 'int' to 'DebounceEvent::TDebounceEventCallback {aka std::function<void(unsigned char, unsigned char, unsigned char, short unsigned int)>}'
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note: DebounceEvent::DebounceEvent(const DebounceEvent&)
 class DebounceEvent {
       ^
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note:   candidate expects 1 argument, 4 provided
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note: DebounceEvent::DebounceEvent(DebounceEvent&&)
/Users/name/Documents/Arduino/libraries/xoseperez-debounceevent-d74ede877b15/src/DebounceEvent.h:39:7: note:   candidate expects 1 argument, 4 provided
button:143: error: no matching function for call to 'std::vector<button_t>::push_back(<brace-enclosed initializer list>)'
             _buttons.push_back({new DebounceEvent(BUTTON2_PIN, BUTTON2_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON2_RELAY});
                                                                                                                                       ^
/Users/name/espurna/code/espurna/button.ino:143:135: note: candidates are:
[... cut ...]
    #define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH`

to
`#define BUTTON1_MODE        BUTTON_PUSHBUTTON
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON`
it compiles fine and I can flash it. Unfortunately the Sonoff Dual R2 doesn't provide an access point after restart so I can't use it. Not sure if this is a side effect of the changed button behavior.

xoseperez, 2018-03-04:
Have you installed the DebounceEvent library? https://bitbucket.org/xoseperez/debounceevent/

skyynet, 2018-03-07:
I have v2.0.0 of DebounceEvent installed. I'm just a user so I don't know what I would have to do to "force the cast in the constructor call". If you take me by the hand I can try.

closed 2018-03-07 by skyynet