xoseperez/espurna #620: Switch Selector in Domoticz

Mattleyu30, 2018-03-03:
Hi, Xose
I'm trying to use the selector button in domoticz with the sonch 4CH with the espurna firmware, but at the time that I had to enter the command http: // or Script: I stopped because I did not find anything on the wiki that told me a script to send from domoticz to espurna for switching off or on.
I had already tried it a long time ago with Espeasy, with a simple http: // command and it worked regularly, but I prefer your firmware much more.
which script can be used to send a command to espurna?

Thanks

xoseperez, 2018-03-04:
You can use the REST API to send commands to ESPurna via HTTP. Check the documentation here: https://github.com/xoseperez/espurna/wiki/RESTAPI

Mattleyu30, 2018-03-05:
thanks xose
I try the REST API, but a question, you say that it is already espurna is already configured for Domoticz, but the switch selector in domoticz, has a single IDX and works with% levels. Does Espurna also manage this type of parameter?

xoseperez, 2018-03-06:
No, the current integration with Domoticz switches only accepts on/off values.

Mattleyu30, 2018-03-06:
ok Thanks Xose

closed 2018-03-06 by Mattleyu30