xoseperez/espurna #622: OBI Powerplug-Adapter (Sonoff S20 type of device)

Geitde, 2018-03-04:
The DIY store OBI in germany is selling a compatible power plug switch for 10 Euro, which is nice as you don´t have to wait for devices from china and as they are selling them in local shops it is guaranteed the quality has been somewhat checked. They can switch up to 3500W, which is at least more as the Dual/Basic/Touch switches can handle, which is nice.

Here is a link to arendst page.

https://github.com/arendst/Sonoff-Tasmota/wiki/Obi-Socket

Using the source below makes Espurna compatible with this power adapter. I bought 3 of them and they work fine incl LED, switch and relay. Thanks to K.H for working it out.

The ready to flash binary using 1.12.3 espurna can be found here: http://www.geit.de/tmp/espurna-1.12.3-obi.bin

Would be nice if the adapter could be added to the regular build system.

There is a little issue with this source. Alexa (WeMo/Belkin International inc. servers) are not detecting them by default. It is strange and seems to be related to some of the id fields. If you flash the normal S20 binary first, search for devices using alexa and flashing the right OBI firmware afterwards it works fine, so it has to be the device string or something else to break the finding process using this source.

Maybe you know what is required to allow proper WeMo detection. Here the source snipped to make it compatible.

Thanks for your hard work on this project.

#if defined(OBI_WIFISTECKDOSE)

// Info
#define MANUFACTURER "EUROMATE"
#define DEVICE "OBI_WIFI_STECKDOSE"

// Buttons
#define BUTTON1_PIN 14
#define BUTTON1_MODE BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON1_RELAY 1

// Relays
// #define RELAY1_PIN 12
// #define RELAY1_TYPE RELAY_TYPE_LATCHED_INVERSE
// #define RELAY1_RESET_PIN 5
// hack: drive gpio12 low and use gpio5 as normal relay pin:
#define RELAY1_PIN 5
#define RELAY1_TYPE RELAY_TYPE_NORMAL
#define LED2_PIN 12 /* DUMMY: exploit default off state for GPIO12=low */
#define LED2_PIN_INVERSE 0

// LEDs
#define LED1_PIN 4
#define LED1_PIN_INVERSE 0
#elif defined(NODEMCU_LOLIN)

xoseperez, 2018-03-05:
Thanks, clever hack for the latched relay, but wasn't it working the RELAY_TYPE_LATCHED_INVERSE type?

xoseperez, 2018-03-05:
Staged for release

Geitde, 2018-03-05:
To my knowledge there is no RELAY_TYPE_LATCHED_INVERSE. That was the basic issue.

xoseperez, 2018-03-06:
True, there is no RELAY_TYPE_LATCHED_INVERSE... You can of course swap the GPIOs but you cannot make them use inverse logic atm.

xoseperez, 2018-03-06:
I've just added the option to dev branch. @Geitde  Maybe you can give it a try?

Geitde, 2018-03-06:
We tried and it fails. It just turns the relay on/off for a blink of an eye.

xoseperez, 2018-03-06:
Do you know the relay type the device has? Datasheet?

Geitde, 2018-03-06:
Well, I will find out tomorrow, when I open them again.

I "killed" one of them in an attempt do reflash them to something older.
OTA Flashing ins no longer possible. The telnet way failed with an instant disconnect and using one of the core versions did the flash, but then failed to connect to wifi.

Geitde, 2018-03-07:
Ah, I remember now. We (a Friend an I) discussed it when hacking the first obi adapter we got.

They relay is not bistable. They used a flip/flop circuit on the board.

xoseperez, 2018-03-12:
I will leave it as it is. Maybe someday we can add a different relay type for this kind of circuit.

closed 2018-03-12 by xoseperez

arthurf1969, 2018-12-11:
Just as an information for you: I originally did not take a closer look at the housing of the device. On the back of the device there is the name EUROMATE (injection molded in the housing - white on white ... silly is). Euromate is part of the OBI brand. Thus my original thought on OBI. But to me (similar to the version 1) this is just a rebranded device from some other chinese brand. In contrast to the first version it does not use a latching relay.