xoseperez/espurna #628: espurnaRegisterLoop with option to define delay between calls

Misiu, 2018-03-06:
Hi Xose,
first of all I'd like to thank You for creating such awesome project!
I'm just starting with Arduino, ESP8266 but Your code is mine of knowledge for me (I'm not native speaker, so sorry it this sentence doesn't make much sense :))

I'm trying to modify Your code a bit with support for SD card.
My idea is to have DS18B20 sensors, read value from them each X minutes and store readings on SD card.
I'm aware I can store current millis and do a diff between current and stored:

```
unsigned long currentMillis = 0;
long previousMillis = 0;
long interval = 1000;

void loop()
{
    currentMillis = millis();
    if (currentMillis - previousMillis > interval)
    {
        previousMillis = currentMillis;
        //save to SD
    }
}
```

but ideally I'd like to be able to call `espurnaRegisterLoop(savetoSd, 300000)`, this would call `savetoSd` every 5 minutes (300 000 millis = 5 minutes).

I'd like to avoid writing SD card as fast as possible because then I would kill it very fast.
What do You think about such functionality? Is there a chance You could add it?

xoseperez, 2018-03-06:
The `espurnaRegisterLoop` method to register loop callbacks has been recently introduced as a first step to add a full task manager, so yes. This is going to be added soon.

Misiu, 2018-03-06:
@xoseperez thank You for reply and awesome news :) Looking forward to this feature.

One more: awesome project 🏆, I'm learning so much from reading Your code.

Misiu, 2018-05-08:
There are some libraries that could be used for that, for example: https://github.com/arkhipenko/TaskScheduler
@xoseperez  what do You think?