xoseperez/espurna #629: [Question] Watchdog support

Misiu, 2018-03-06:
I have a newbie question: Do You support watchdog timer?
I'm trying to include SD card support so I'd like to avoid any potential (caused by me) failures.
One thing that I can think of is software Watchdog, that should reset my Wemos after it stops responding for a while.
Ideally there should be an option to use software and hardware watchdog (like this for example: https://github.com/dzzie/home_automation/tree/master/LongWatchdog)

it this is already supported sorry for all the trouble.

xoseperez, 2018-03-06:
The ESP8266 has a hardware watchdog and the SDK a software watchdog. Basically, they will reset the board when it gets stalled. There's no additional watchdog in the firmware. 

I can picture an additional hardware watchdog like something that would monitor a certain GPIO that should output some kind of square wave and reset the board if it stops doing it.

Misiu, 2018-03-07:
@xoseperez I didn't knew about "automatic" watchdog in ESP8266.
You are right about external hardware watchdog - if it won't get signal from certain GPIO then it resets the board.
I was thinking about adding such watchdog, but I guess I must read more about ESP8266.

xoseperez, 2018-07-01:
Closing since there has been no feedback for several months. Feel free to reopen if necessary.

closed 2018-07-01 by xoseperez