xoseperez/espurna #630: Was support for HomeAssistants MQTT_Json implemented?

jonny190, 2018-03-06:
I can see this was brought up in #235  

But from what ive read the auto discover was fixed with no mention of the comparability to use home assistants MQTT_Json (https://home-assistant.io/components/light.mqtt_json/)

I'd live to see the transition feature be implemented over the current ON/OFF functionality

xoseperez, 2018-03-06:
There are no plans to support MQTT_JSON atm. I will flag this issue as an enhancement. If anyone wants to contribute it will be welcomed.

Skaronator, 2018-04-09:
I'll take a look since I need transitions as well.

Skaronator, 2018-05-03:
@xoseperez 
I gave this a try. I got it working but I don't like it how it currently is implemented. The main issue is that I need values from light.ino (RGB, Brightness, Color Temp) in relay.ino (ON/OFF) and vise versa.

I'm somewhat lost. The MQTT Topic is similar to `/relay/#/set` so you could switch multiple relays. Eg. `H801-202/json/0/set` and `H801-202/json/1/set` but it feels like that `light.ino` is only designed for a single switch?

Moving the data from `light.ino` to `relay.ino` works fine. I got the `_lighMQTTJSONState()` function in `light.ino` which "sends" the data to `relay.ino`. But I don't know how I should move data the other way around, from `relay.ino` to `light.ino`.

To be more specific I need this value for relay X:

https://github.com/Skaronator/espurna/blob/0f3954ab23100462ab67159f02d3f2ff5c7b6647/code/espurna/relay.ino#L642

in `light.mqtt`

https://github.com/Skaronator/espurna/blob/0f3954ab23100462ab67159f02d3f2ff5c7b6647/code/espurna/light.ino#L620-L651

I don't even know which relay I should use. `light.ino` feels like it is written for a single relay?

Here is the current implementation (also check the whole MQTT_JSON branch):
https://github.com/Skaronator/espurna/commit/0f3954ab23100462ab67159f02d3f2ff5c7b6647

Currently missing:
 - Transition time (should be very easy)
 - Send update on color change (see above, lightMQTT function in light.ino)

xoseperez, 2018-05-03:
I guess the trouble is caused by pretending to use the relay module with a light device. It's not your problem, is inherited by ESPurna architecture but I feel this is wrong. The relay module is a first class citizen since everything started with a relay, wifi and MQTT. But there is no reason there should be a relay here.

At the same time the relay is very useful since there are a lot of features on top of it (pulse, scheduling,...). But maybe the light module should have it's own on/off routines...

Skaronator, 2018-05-04:
Would be a good point to directly support multiple light switches, maybe even dual RGB or dual CCT strips.

Modifying the MQTT command topics to include a switch id - eg.:
```diff
- {root topic}/mired/set
+ {root topic}/0/mired/set
- {root topic}/rgb/set
+ {root topic}/0/rgb/set
```
So we could set the color temperature of the first CCT strip with `{root topic}/0/mired/set` and the second CCT strip with `{root topic}/1/mired/set`. Same for on and off `{root topic}/0/switch/set` and `{root topic}/1/switch/set`.

Would be cool to configure the channels directly from the Webinterface (or while compiling). For example:

Channel | Type
------------ | -------------
0 | warm white 1
1 | cold white 1
2 | normal channel
3 | warm white 2
4 | cold white 2

or dual RGB strip over 6 channels (if there is a device? or somebody build its own)

Skaronator, 2018-05-05:
I've updated the code in my fork. The `light.ino` file has now its own MQTT Topic called "switch". I also change the MQTT Topic of the other light commands in order to support multiple light switch in the future. (And we don't need to have 2 breaking changes between 2 updates)

```diff
- red is the old topic
+ green the new topic

- H801-201/relay/0/set
+ H801-201/switch/0/set
- H801-201/brightness/set
+ H801-201/brightness/0/set
- H801-201/rgb/set
+ H801-201/rgb/0/set
- H801-201/mired/set
+ H801-201/mired/0/set
```

MQTT_JSON support works. The topic is `H801-202/json/0/set`. The HA config would look like:

```
  - platform: mqtt_json
    name: H801_202_JSON
    qos: 0
    state_topic: H801-202/json/0
    command_topic: H801-202/json/0/set
    availability_topic: H801-202/status
    payload_available: 1
    payload_not_available: 0
    brightness: true
    rgb: true
    color_temp: true
```

Code is in the MQTT_JSON branch: https://github.com/Skaronator/espurna/commits/MQTT_JSON

Known Issue:
 - The old `H801-201/relay/0/set` should be ignored when there are no relays. (Eg. light only)
 - Transition not implemented yet.

What do you think @xoseperez?

xoseperez, 2018-05-08:
I'm worried about the topic naming change since it breaks backwards compatibility. Maybe it should only change if there is indeed more than one light switch.

Also we should be aware than detaching the relay module from the light module we are losing some features, like pulses or MQTT sync...

commit 4d157ccd (2021-01-25): ha: discovery refactoring

- retain setting
- generate topic and message in the discovery object instead of
creating them all at once
- re-implement relays, lights and sensors discovery
- rework name normalization, use relative magnitude indexes
- replace original lights code with schema: json, subscribe to a special
topic to handle lights instead of using the common API
(still wip - colors, hs, temperature settings)

resolve #630
resolve #1564
resolve #2403
closed 2021-01-25 by mcspr