xoseperez/espurna #631: Simple Button/Switch

pidiet, 2018-03-06:
Hello!

Great, great, GREAT software!!! =) Thank's so much!

As with the latest build there is a core release (#557 ) included. I wounder what's needed to use this and a wemos d1 mini with a button shield to have a 'smart button' for home assistant.

[image]

I guess everything there already because I use a wemos with the relay shield and the button shield and it just works without modifications!

[image]

Question now: How to use the button without the hardwired relay but just as an mqtt event?

lobradov, 2018-03-06:
I’m just experimenting with similar setup - have 6 buttons and no relay. 
Don’t have it working yet, but will post progress once I have something.

lmarlow, 2018-03-06:
I added my own configuration in platformio.ini that adds `-DBUTTON2_PIN=16` to the `build_flags`. This will send MQTT messages once you setup MQTT. I use it for a reed switch connected to my garage door to tell me when it is closed.

xoseperez, 2018-03-06:
Buttons send MQTT messages upon any event. The topic is `<root>/button/#` and the payload will be the event type. Check the wiki page [here](https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#buttons-and-switches) to know the codes of the button events. If you don't specify a BUTTON#_RELAY then the button will not be linked to any relay.

lobradov, 2018-03-09:
I can confirm this works as expected. 

Here's my setup (from my `custom.h`):

```
#if defined(WEMOS_6BUTTONS)

    // Info
    #define MANUFACTURER        "Wemos D1 mini"
    #define DEVICE              "Wemos + 6 button shield"

    // Buttons
    #define BUTTON1_PIN         5
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH 

    #define BUTTON2_PIN         4
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON3_PIN         0
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON4_PIN         14
    #define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON5_PIN         12
    #define BUTTON5_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH 

    #define BUTTON6_PIN         13
    #define BUTTON6_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1
#endif
```

Lessons learned: 
- Always use (hardware) pull up resistor. I used a SIL8 10k resistor network to simplify packaging. Without it, buttons were floating and giving me almost random results.
- Espurna will send `1` when button is pressed and `2` when it's depressed, so use your home automation software to react to one of the events. I initially configured the system to toggle remote relays on on any value, leading to very fast on/off of relays as there will always be 2 events per press.
- Tweak `BUTTON_DBLCLICK_DELAY`, `BUTTON_LNGCLICK_DELAY` and `BUTTON_LNGLNGCLICK_DELAY` if you plan using it. 

Hope this helps. 

Please close the issue if you are happy with the answer, but feel free to keep it open if you require further help.

rradar, 2018-03-22:
Hey!

thank's great. I look for a easy solution to have a simple button like this! It should connect to mqtt to be able to trigger things in home assistant.

@lobradov sadly I can't follow your quite technical explanations. I used to flash the binaries from the release section. I think you building your own? Is it necessary or could this also achieved with a binary and some settings as well? When not would you mind elaborate your tutorial a little bit so i can reproduce it?

Or maybe it could be even integrated as a own build? like the wemos-d1-mini-relayshield a wemos-d1-mini-buttonshield?

pidiet, 2018-03-26:
Yea! wemos-d1-mini-buttonshield release sounds awesome =)

closed 2018-06-04 by xoseperez