xoseperez/espurna #633: UART_MQTT_SUPPORT causes crash when connect over telnet

AlbertWeterings, 2018-03-06:
Never seen this before. Until now I was unable to find why this happens:

[049050] [INIT] BUILD_FLAGS: 
[049050] [INIT] BOARD: NODEMCU_LOLIN
[049050] [INIT] SUPPORT: BROKER DEBUG_TELNET MDNS_SERVER NTP TELNET UART_MQTT WEB

[049051] [INIT] Last reset reason: External System
[049051] [INIT] Settings size: 1315 bytes
[049052] [INIT] Free heap: 25768 bytes
[049053] [INIT] Power: 2767 mV
[049053] [INIT] Power saving delay value: 10 ms
[049053] 
[049053] [WIFI] MODE STA -------------------------------------
[049053] [WIFI] SSID  XXXX
[049053] [WIFI] IP    192.168.1.164
[049054] [WIFI] MAC   5C:CF:7F:D5:1E:4B
[049054] [WIFI] GW    192.168.1.1
[049054] [WIFI] DNS   8.8.8.8
[049054] [WIFI] MASK  255.255.255.0
[049054] [WIFI] HOST  ESPURNA_D51E4B
[049054] [WIFI] BSSID B0:B2:DC:6A:92:B4
[049054] [WIFI] CH    11
[049055] [WIFI] RSSI  -47
[049055] [WIFI] ----------------------------------------------
[049055] [DEBUG] Crash at 7000 ms
[049055] [DEBUG] Reason of restart: 2
[049055] [DEBUG] Exception cause: 3
[049055] [DEBUG] epc1=0x4010011d epc2=0x00000000 epc3=0x00000000
[049055] [DEBUG] excvaddr=0x40032730 depc=0x00000000
[049055] [DEBUG] >>>stack>>>
[DEBUG] 3fff1880: 00000000 3fff1910 4023cda1 3ffefda0 
[DEBUG] 3fff1890: 3fff08cc 00000095 00000095 4010020c 
[DEBUG] 3fff18a0: 3ffe83d5 00000000 3fff22dc 4010068c 
[DEBUG] 3fff18b0: 3fff18f0 00000000 3ffe83d5 4010731c 
[DEBUG] 3fff18c0: 4023c4ba 3fff22dc 3fff22dc 4023b183 
[DEBUG] 3fff18d0: 4023d2de 3fff22dc 3fff22dc 4010020c 
[DEBUG] 3fff18e0: 40214fb1 3fff0578 3fff1910 4010068c 
[DEBUG] 3fff18f0: 00000000 00000000 0000000c 4021fce0 
[DEBUG] 3fff1900: 3fff08cc 000000ab 000000ab 4010020c 
[DEBUG] 3fff1910: 00000000 00000000 3fff1960 4010068c 
[DEBUG] 3fff1920: 40105a64 00202e91 3fff2734 402208b8 
[DEBUG] 3fff1930: 3fffdad0 3fff014c 00000000 4020e9cb 
[DEBUG] 3fff1940: 401057d6 002070c0 3fff0504 3fffdad0 
[DEBUG] 3fff1950: 4023c4ba 3fff0578 00000001 3ffe83d5 
[DEBUG] 3fff1960: 40214a1c 3fff060c 3fff0578 40216e52 
[DEBUG] 3fff1970: 3ffefb20 3fff0[060001] [MAIN] System OK

xoseperez, 2018-03-06:
When UART_MQTT_SUPPORT is enabled DEBUG_SERIAL_SUPPORT and TERMINAL_SUPPORT are disabled. In this mode whenever you connect via telnet it will dump the info and the latest crash dump. That does not mean the system is crashing. 

Actually, you can see a "System OK" at 60s while the crash dump was at 49s. Just check you have connectivity.

But maybe I should clarify the message to avoid this type of confusion...

xoseperez, 2018-03-06:
Changed the copy with 70f82aa. Also forced to clear crash info after displaying it.

AlbertWeterings, 2018-03-06:
Damn.. I did not notice this... Was already trying to get rid of this crash dump ;-)

AlbertWeterings, 2018-03-06:
Yep problem solved. Great, one additional question I have read somewhere you are going to make more settings changeable in the web interface or console. Can you add enable/disable UART_MQTT_SUPPORT and its baud rate to the settings in the web interface also at that time?

xoseperez, 2018-03-06:
The idea was to be able to configure basic hardware (relays, buttons, LEDs) and sensors.

xoseperez, 2018-03-12:
Staged for release

closed 2018-03-12 by xoseperez