xoseperez/espurna #634: Power cycle delay for HID lamps and motors

sashimanu-san, 2018-03-06:
Some mains AC loads require a cool down period of several minutes after shutting down before powering on again, e.g. high intensity discharge lamps won't reignite, and a refrigerator unit may overheat. 

Since the firmware has a software clock and NTP support, it is possible to cover some cases pretty easily:
- manual, remote or scheduled relay operation: 
  - when real time is available &mdash; 
      - just write the power off timestamp _(ts)_ along with relay state to permanent memory, and don't power on until wall clock is _ts+dt_.
  - when no real time is available and the unit doesn't suddenly reboot in the meanwhile
    - count the seconds
   - unit reboots gracefully by user request
     - write a flag to permanent memory. Upon boot, read the flag and wait _dt_ seconds if present, then power on the relay if needed.
 
A tough case is the following:
- recycling after a sudden unit reboot due to a power loss or other cause. Implementing this feature in software will require writing heartbeat to permanent memory every so often which may raise flash wear issues.

Any thoughts?

xoseperez, 2018-03-06:
If the unit reboots due to a watchdog reset or a bug there is the chance to write to EEPROM before existing. If it reboot due to a power loss or hard disconnection there is no way unless what you suggest, which I wouldn't do.

sashimanu-san, 2018-03-06:
I think I've found a solution for this case:

**If** the unit is restarting for reasons suspicious and time unknown
**and** the stored relay state is ON
**then** wait for the delay to expire
**however** if the relay is commanded ON by a button press or network request during this period, power on immediately.