xoseperez/espurna #635: Sonoff Pow without power consumption sends values greater than zero

Maergsche, 2018-03-06:
The Sonoff Pow transmits power values up to 8 watts even when no electronic devices are connected. It does not matter if the relay is closed (on) or open (off).

I using Version 1.12.4 and services like Alexa and telnet are deactivated. Only MQTT is in use. 

[image]

AlbertWeterings, 2018-03-07:
The sonoff pow shout not measure any energy when the relay is in off position at that point it short circuit the resistor used to measure energy so the voltage difference on HLW8012 pin 2(vip) and 3(vin) shout be 0volt. The only difference I see in other designs is that sonoff has 100nf capacitors to ground on pin 2 and 3 and in all other designs i see 33nf. It might be possible the HLW8012 is picking up noise.

xoseperez, 2018-03-11:
Sensor values are filtered using a median filter to remove spurious values but apparently is not enough as the HLW8012 is picking up noise all the time.

Tardymo, 2018-03-11:
Yes, I can confirm that POW is reporting wattage spikes even when relay is off. I was hoping to trigger off wattage when connected device turns on, so I decreased reporing interval, but it is unuseable with very short intervals as I see up to 30W spikes.

AlbertWeterings, 2018-03-13:
The pow design has a ceramic capacitor at the AMS1117 this might result in it oscillating @xoseperez In your design with the HLW 8012 what kind of capacitors did you use and do you see the same issue?

xoseperez, 2018-03-15:
I use tantalum capacitors at the ASM1117 but it does have noise nevertheless. But they are usually ON so I cannot tell if everything is noise or there is actually some residual power usages (LEDs and so).

AlbertWeterings, 2018-03-31:
@Maergsche Can you please check if your SonOff pow has a CSE7759 power measuring chip or a HLW8012?

closed 2018-08-13 by stale[bot]