xoseperez/espurna #636: Set the default branch to `dev`

lobradov, 2018-03-07:
@xoseperez, I see many people (myself included) working against `master` branch, simply because it's set as default. Once PR is submitted, you need to ask people to rebase (or rebase yourself) which is not always easy task. 

How about setting `dev` to be default branch?

lobradov, 2018-03-08:
... also, on the same note, it would be good to have different `.travis.yaml` for `master` and for `dev` branch. 

The one in master would run a full build, with all 50+ images, while the one in dev would just compile one "all-sensors" big dummy image to check if something is broken.

xoseperez, 2018-03-08:
I've been thinking about this also. My concern is that people will start building the dev branch now which might not be always stable. Of course, I can use specific branches for each new feature, which I sometimes do.

About the .travis.yml, absolutely. Having different contents is tricky. The only approach I know and I use is a pre-commit hook like the one I use in ESPurna to changed the badges in the README file. Do you know of a better way to do it?

lobradov, 2018-03-08:
I've seen people adding `.travis.yaml` to `.gitignore` not to accidentally merge it from dev to master.
 
Other option is to have same `.travis.yaml` but evaluate `${TRAVIS_BRANCH}` and decide what do to, aka: 
```bash
if [[ ${TRAVIS_BRANCH} = "master" ]]; then 
  ./build.sh
else
  ./build.sh dummy-image
fi
```

As for dummy-images, take a look at #643, I gave it a try and seems I already caught something that doesn't build properly.

lobradov, 2018-03-08:
On building from `dev` and expecting stable release... 

I think folks either download ready-made images (which are build from `master` anyway), or decide to build their own because they want some of the latest-and-greatest stuff and are ready to accept some risk. 

I'll be happy to document `git clone https://github.com/xoseperez/espurna.git -b master` approach for building stable images if you think it might help, but default branch should really point to a base branch for PRs, as documented on Settings -> Branches -> Default branch: 

> The default branch is considered the “base” branch in your repository, against which all pull requests and code commits are automatically made, unless you specify a different branch.

xoseperez, 2018-03-09:
Just changed the default branch, added the Travis branch check code to the build script and added a message in the README file about the change.

closed 2018-03-09 by lobradov