xoseperez/espurna #637: Web interface not accessible on the core releases

pidiet, 2018-03-07:
Hello!

Just tried to get a core release (the 4MB and the 1MB version) running on a wemos d1 mini. The Wifi comes up and I can connect to it (tried from 2 different devices). Problem is that the web interface is never accessible.

Just to be sure I tried flashing with esptool.py (always worked. checksum OK) and also directly inside the web interface with a working build (wemos relay shield). No web interface - never.

Is it intentional or is it a bug?

Geitde, 2018-03-07:
Yeah, I ran into the same trap, when I wanted to go back to a working version on my OBI plug and found out that I only get "the firmware is to big" information.

Wifi seems to be fine.

thebradleysanders, 2018-03-07:
Please View: https://github.com/xoseperez/espurna/wiki/TwoStepUpdates

About ESPurna Core
The ESPurna core image is basically a stripped-off version of the ESPurna firmware without several features that usually come bundled in. So, aside from the features that are not enabled by default, you don't have either:

Alexa support
Domoticz support
Home Assistant auto-discover
I2C support
MQTT
NTP sync
Web interface
HTTP API
Sensors
Scheduler
Thinkspeak support
But you do have:

mDNS
OTA
Telnet
Terminal
WiFi

Geitde, 2018-03-07:
Thanks for the hint. I did not know that. telnet ota command worked for me to recover one device from "core" mode, but it makes sense this way.

pidiet, 2018-03-07:
OK. The **missing web interface is intentional**. Thank's for the clarification!

closed 2018-03-07 by pidiet