xoseperez/espurna #638: RF-Bridge: Change dummy relay count

ctandi, 2018-03-07:
Hello,

I'd like to add 35 devices to the RF-Bridge. If I change the dummy_relay_count to more than 24, the "RF" tab in the webinterface disappears. I've already tried disabling functions and extended the EEPROM size to 16 Kilobytes with no luck. Is there anything else I have to change in order to make it work or is it simply impossible due to RAM/Flash limitations. 

Thanks in advance!

xoseperez, 2018-03-11:
The EEPROM size is defined by the flash layout, not easy to change... if you really need that many devices I'd suggest you to use the RFBridge as an MQTT-to-RF bridge and send codes from somewhere else (Node-RED for instance). You won't have to learn and assign different codes to virtual relays in ESPurna, instead you can just sniff the codes in `<root topic>/rfin` and send them to `<root topic>/rfout/set`.

xoseperez, 2018-07-01:
Closing. Please, feel free to reopen if necessary.

closed 2018-07-01 by xoseperez