xoseperez/espurna #641: Error flashing Electrodragon Wifi IoT Relay Board SPDT

skyynet, 2018-03-07:
Received an Electrodragon board and wanted to flash it. Get following error:
```Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DIO, 1M (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Archiving built core (caching) in: /var/folders/7t/gtn_36_50gd85n5dhtlv4k7m0000gn/T/arduino_cache_465857/core/core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_dio,FlashSize_1M0,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_5d05bc9e751b6a00274f4ca4a6717e01.a
Der Sketch verwendet 490775 Bytes (47%) des Programmspeicherplatzes. Das Maximum sind 1023984 Bytes.
Globale Variablen verwenden 42972 Bytes (52%) des dynamischen Speichers, 38948 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```

Do I have to change anything in Arduino IDE? Flashed Sonoffs before and now I only changed the board in the arduino.h config file.I hold down BTN2 while booting.

[image]

thebradleysanders, 2018-03-08:
Try switch the RX/TX, there may be an issue with the board silk screen.

thebradleysanders, 2018-03-08:
What board layout are you using? Generic esp8266?

skyynet, 2018-03-08:
I use Generic esp8266. After changing tx-rx rx-tx back and force it suddenly worked. These boards seem to be a little bit touchy. Closing the issue.

closed 2018-03-08 by skyynet