xoseperez/espurna #644: relay option sonoff basic

bwired-nl, 2018-03-08:
Hi 
I want to use the free GPIO 14 to mount a motion sensor.
this works OK
now the relay is switching both on gpio14 and the pushbutton

is it possible to not switch the relay on Gpio14 but only send the MQTT message
can this be done in a setting?

xoseperez, 2018-03-11:
GPIO14 is defined by default as a button and linked to the relay. You can just unlink the button commenting or removing this line:

https://github.com/xoseperez/espurna/blob/5f2b22d06319b5d8169ccbfdfec280d9d2d47c04/code/espurna/config/hardware.h#L213

xoseperez, 2018-07-01:
Closing. Please, feel free to reopen if necessary.
Check https://github.com/xoseperez/espurna/projects/2 to know when this will be possible using the web UI

closed 2018-07-01 by xoseperez