xoseperez/espurna #648: RFBridge buzzer

myounges, 2018-03-09:
Hello first of all would like to thank you for your great work, my hat goes off for you.
I have an enhancement suggestion unless I'm not aware of possible ways to do it.

It is regarding the buzzer the RFBridge has included, is there any way to enable that an mqtt message can enable the buzzer for a second or so? Lets say that the buzzer is connected to GPIOX we can assign the mqtttopic/relay/4/set -->1/0 or similar. I believe it is not that easy as it doesn't seem to be connected to a GPIO directly, but it is worth asking.

I use the RFbridge to enable or disable an RF alarm and would be interesting to beep once the alarm is enabled.

thanks!
Michael

myounges, 2018-03-09:
Me again...

Doing some research I see the buzzer is connected to the EFM8BB1 chip. 
[image]

Maybe through the UART pins, this can be done?

https://www.itead.cc/wiki/images/3/3e/Sonoff_RF_Bridge_433_Schematic.SCH.pdf

Cheers,

xoseperez, 2018-03-11:
Nom the buzzer is controlled by the EFM8BB1 and there is no command in the protocol to drive it from the ESP8266.

myounges, 2018-03-13:
Ok, thank you very much!

Portisch, 2018-03-13:
Beep will be supported with the new EFM8 firmware:
https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/fc5774fb956d7b7610b2573d324393bdb2b5b1d2

AlbertWeterings, 2018-03-13:
As @Portisch has included the beep feature in his firmware I will later create a pull request to add the feature to ESPURNA. My idea is to add a MQTT topic where you can send the number of beeps you want to hear to. If anyone has a better or other idea to implement this then please share your thoughts.

myounges, 2018-03-14:
That would be great! Mqtt topic would be perfect!
thanks guys!

closed 2018-08-13 by stale[bot]