xoseperez/espurna #654: WEB_EMBEDDED or nothing :)

PieBru, 2018-03-10:
Hi, is there an easy way to build and serial upload with WEB_EMBEDDED disabled?

In most cases the transition to WEB_EMBEDDED has been positive, but the old spiffs way may be useful in some circustances, for end users as well as for debuggers, i.e. hopefully to help sorting out #652 and sonoffsc issue#29.

Thanks,
Piero

mcspr, 2018-03-10:
Not actually related to build process, but you can debug with local web server. If you got python installed, it got you covered.

go to **code/html** and run `python3 -m http.server 8888` or `python2 -m SimpleHTTPServer 8888`
then in browser go to http://localhost:8888
and in web inspector console run `connect("10.1.2.3")`, where 10.1.2.3 is IP of real device.

PieBru, 2018-03-10:
Thanks anyway @mcspr, that's a nice trick.
But what's the advantage over using the web inspector while connected directly to the ESP device?

mcspr, 2018-03-11:
@PieBru Yeah, I misinterpreted that as a question about modifying html and other resources in-place on device.
Reason for doing that local thing is because you can run device with WEB_EMBEDDED=0, still use web interface and inspector will show actual source code. So index page flow can be modified - hooking into ws, functions, variables from the start.

xoseperez, 2018-03-11:
I think the title of the issue is quite right :smile: 
Debugging the web interface and the communication with the device can be done like @mcspr  says. Actually, you don't even need a web server, just open the `index.html` file with your browser and use web inspector to call the `connect` method with the IP of the physical device.
The only useful alternative to WEB_EMBEDDED I can think of right now is not having the web code in the devices at all and create some sort of "managing platform" to control them all.