xoseperez/espurna #656: [Solved] Debug over MQTT, instead of problematic UDP

PieBru, 2018-03-10:
Hi, I need to debug wirelessly and I’m looking to an alternative to UDP, which currently suffers some issues as #651, #652, #654.
Quite the opposite to UDP, MQTT config is already definable in the WEB and console UI.
So here is my experimental simple proposal, a mqttSendDebug() in mqtt.ino and a DEBUG_OVERMQTT_SUPPORT in debug.ino and general.h
The next step will be sending embedis console commands via MQTT ;)
Sorry for not pulling in /dev, but I’m still not confident with git :(

debug.ino, at the end of _debugSend()
```
    #if DEBUG_OVERMQTT_SUPPORT
        mqttSendDebug(message);
    #endif
```

general.h
```
// -----------------------------------------------------------------------------
// Debug over MQTT
// MQTT debug is more flexible than UDP as the sender doesn’t need to define who is the final receiver(s),
// but the MQTT underlying stack is more complex than UDP and when stressed it may suffer some issues.
// Let's assume the MQTT library, which is used more extensively than UDP, is mature and reliable.
// While the easiest way to debug pre-connection phases is via serial interface, which is available
// on most (but not all) esp8266-based things, debug over MQTT may be useful to debug application-level
// on already connected things, even when a serial wire isn’t available, of course if the debug
// doesn’t concern the wireless connection or MQTT itself.
// At the moment, this works on ESP8266 in STA mode but not in AP mode, so the ESP must be connected to a wifi LAN.
// -----------------------------------------------------------------------------

#define DEBUG_OVERMQTT_SUPPORT  1           // Send console messages to the connected MQTT broker, using mosquitto CLI use:
                                            // mosquitto_sub -v -h <my_mqtt_broker> -t '$DBG/#'
#if DEBUG_OVERMQTT_SUPPORT
#undef MQTT_SUPPORT
#define MQTT_SUPPORT            1           // Of course, MQTT support is required to play this game
#endif

#define DEBUG_OVERMQTT_HEAD     "$DBG/"     // The topic string to put before the host name, terminated with a slash
#define DEBUG_OVERMQTT_TAIL     ""          // An empty string, or a slash followed by the topic string to put before the host name
```

mqtt.ino, just before mqttSendRaw()
```
#if DEBUG_OVERMQTT_SUPPORT
void mqttSendDebug(const char * message) {

    if (!_mqtt.connected()) return;

    // Prepare topic
    char topic[32];
    snprintf_P(topic, sizeof(topic), PSTR("%s%s_%06X%s")
        , DEBUG_OVERMQTT_HEAD, APP_NAME, ESP.getChipId(), DEBUG_OVERMQTT_TAIL);

    // Discard terminating control chars
    int msglen = strlen(message);
    while (msglen > 0 && message[msglen-1] < ' ') msglen--;

    // Send
    #if MQTT_USE_ASYNC
        uint16_t packetId = _mqtt.publish(topic, 0, false, message, msglen);
        //if (packetId == 0) Serial.printf("[MQTT] Error sending %s => %s\n", topic, message);
    #else
        bool rc = _mqtt.publish(topic, message, false);
        //if (rc != true) Serial.printf("MQTT] Error sending %s => %s\n", topic, message, msglen);
    #endif

}
#endif  // DEBUG_OVERMQTT_SUPPORT
```

It works quite flawlessy and I found it useful.
Piero

xoseperez, 2018-03-11:
I'd prefer not to use MQTT for that. Maybe you can use the new DEBUG_WEB_SUPPORT, I hope you will find it useful... it's enabled by default in the dev branch, so you can build and check the logs right in the web UI.

PieBru, 2018-03-12:
Yes, it may seem crazy, but it solved me a problem when UDP failed.
I will keep it in my dev survival kit :) default disabled so it costs nothing.
DEBUG_WEB_SUPPORT seems promising, will try it ang give a feedback in the next days.
Thanks,
Piero

noname000git, 2018-03-19:
I also think it can be useful when you want to debug remotely without having to deal with port redirection. In this case connecting mqtt with embedis would be also interesting.

closed 2018-05-02 by xoseperez