xoseperez/espurna #661: Homeassistant and Alexa integration, double entries in HA?

WayneManion, 2018-03-11:
I have a bunch of Espurna devices scattered around my house. I enabled Alexa support on most of them so I can control them with voice commands. 

I recently experimented with Homeassistant and the discovery feature in the new Espurna release. It looks like all of the Espurna devices with Alexa integration enabled show up in the HA web interface twice, and only one of them works. I think HA is picking them up as Wemo devices (which are the ones that don't work) and once again through the MQTT auto-discovery feature. 

How can I Alexa integration work on my Espurna devices, but not have them show up as Wemo switches in HA?

swilson86, 2018-03-11:
I have an identical problem with Alexa, but instead of HA I am using node-red. Something is fishy with Espurna as only Espurna devices are entered twice.

xoseperez, 2018-03-11:
ESPurna uses the fauxmoESP library that emulates Wemo devices. The way it emulates them make it answer to Home Assistant discovery requests and duplicates them. But my experience is that both switches work (in home assistant).
@swilson86 you are not using Home Assistant at all? What is your set up?

swilson86, 2018-03-11:
@xoseperez My setup is this: 
1) I program Sonoff Basic, POW, and AiLight's and flash them using your firmware and the Expressif program Expressif Download Tool. 
2) I login using Chrome on Windows 10 and setup a password, and rename the device to "Bar Light" for example. I assign a fixed IP address using the MAC address of the device and then configure it to connect to my WiFi router. I also setup the MQTT section. At the end of all this it works perfectly.
3) I then ask Alexa to discover devices and then Alexa will ALWAYS make a double entry in the alexa.amazon.com page. One with Wemo Switch, another with either Wemo Plug (and sometimes Belkin International". I delete all the duplicates leaving only Wemo Switch as a valid entry. Works okay.
4) I also use node-red and the excellent Alexa Home Skill to group Espurna devices. ie: "Alexa All Bar Lights off" (which successfully turns on/off three Espurna devices). 
5) The Alexa duplication bug is not node-red related as the duplication issue begins way before any node-red programming. It happens right after flashing a new Sonoff/AiLight device and immediately after device discovery by Alexa is the source of the issue. Issue #340 has more details.

swilson86, 2018-03-11:
[image]
This is an example of how I flash a Sonoff Basic.

mcspr, 2018-03-12:
@WayneManion You can use *ignore* key of the [discovery](https://home-assistant.io/components/discovery/) component.
*configuration.yaml:*
```yaml
discovery:
  ignore:
    - belkin_wemo
```

@xoseperez By the way, due to communication issue, Home Assistant cannot control discovered espurna wemo devices. [fauxmoESP 46 (comment)](https://bitbucket.org/xoseperez/fauxmoesp/issues/46/cannot-use-setbinarystate-with-pywemo#comment-43686362)

WayneManion, 2018-03-12:
mcspr, I thought about that, but I have a couple of authentic Wemo devices in my setup. 

If the emulated Wemo devices were operable through HA, that would be great. 

HA appears to have an upper limit of Wemo devices (14 maybe). Alexa used to have a similar limit, but at some point the limit was raised or removed entirely.

mcspr, 2018-03-12:
@WayneManion see my sneaky edit. Plus HA allows to manually list wemo devices when they are known to have same ip addresses via reservation:
```yaml
wemo:
  static:
    - 1.2.3.4
    - 1.2.3.5
...
```

Have you tried listing real ones like this?

closed 2021-10-03 by mcspr