xoseperez/espurna #662: Sonoff T1 EU 2-gang random button 0 events, then reboots, continuously

dote78, 2018-03-11:
Hi, I have been trying to set up 2 of these Sonoff T1 EU 2-gang, and they both do the same. The flashing does not give any errors, but after restarting, button #0 starts reporting presses and events (watching the console) and then it reboots, continuously. Sometimes it works ok for 1 minute, others for 5...

I've even been able to connect to my wifi and mqtt server, but keeps rebooting itself.

I flash with these commands (first erase then flash after booting again into flash mode)

I have also tried firmware 1.12.3 just in case, with the same result

```
esptool.py --port COM3 --baud 115200 write_flash -fs detect -ff 26m -fm dout --verify 0x00000 espurna-1.12.4-itead-sonoff-t1-2ch.bin

esptool.py v2.3-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0321
Compressed 464576 bytes to 338323...
Wrote 464576 bytes (338323 compressed) at 0x00000000 in 29.9 seconds (effective 124.2 kbit/s)...
Hash of data verified.

Leaving...
Verifying just-written flash...
(This option is deprecated, flash contents are now always read back after flashing.)
Flash params set to 0x0321
Verifying 0x716c0 (464576) bytes @ 0x00000000 in flash against espurna-1.12.4-itead-sonoff-t1-2ch.bin...
-- verify OK (digest matched)
Hard resetting...
```

Then all seems to be ok, looking at the terminal, but then, without touching anything, button #0 lits, the console gives this messages, and then reboots

`[056420] [BUTTON] Button #0 event 1
[056421] [RELAY] #0 scheduled ON in 0 ms
[056431] [RELAY] #0 set to ON
[057432] [RELAY] Saving mask: 1
[060001] [MAIN] System OK
[066936] [BUTTON] Button #0 event 5`

Other times is event 4

`[051611] [BUTTON] Button #0 event 1
[051613] [RELAY] #0 scheduled ON in 0 ms
[051625] [RELAY] #0 set to ON
[052626] [RELAY] Saving mask: 1
[060009] [MAIN] System OK
[061375] [BUTTON] Button #0 event 4`

and then the reboot code, giving that strange "Hardware button" as a reason for last reboot. Full sequence (twice)

`[000833] [MDNS] OK
[001664] [RELAY] Saving mask: 0
[060004] [MAIN] System OK
[103793] [BUTTON] Button #0 event 1
[103794] [RELAY] #0 scheduled ON in 0 ms
[103805] [RELAY] #0 set to ON
[104806] [RELAY] Saving mask: 1
[114652] [BUTTON] Button #0 event 5

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000439] 

[000439] [INIT] ESPURNA 1.12.4
[000439] [INIT] xose.perez@gmail.com
[000439] [INIT] http://tinkerman.cat

[000440] [INIT] CPU chip ID: 0xA12C99
[000442] [INIT] CPU frequency: 80 MHz
[000445] [INIT] SDK version: 1.5.3(aec24ac9)
[000449] [INIT] Core version: 2_3_0
[000452] [INIT] Core revision: 
[000455] 
[000456] [INIT] Flash chip ID: 0x144051
[000459] [INIT] Flash speed: 26000000 Hz
[000463] [INIT] Flash mode: DOUT
[000465] 
[000466] [INIT] Flash sector size:     4096 bytes
[000471] [INIT] Flash size (CHIP):  1048576 bytes
[000475] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000481] [INIT] Firmware size:       464576 bytes /  114 sectors
[000486] [INIT] OTA size:            561152 bytes /  137 sectors
[000492] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000498] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000503] [INIT] Empty space:          16384 bytes /    4 sectors
[000509] 
[000510] [INIT] BUILD_FLAGS: -g -DMQTT_MAX_PACKET_SIZE=400  -Wl,-Tesp8266.flash.1m0.ld -DITEAD_SONOFF_T1_2CH
[000520] [INIT] BOARD: ITEAD_SONOFF_T1_2CH
[000523] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000631] [INIT] Last reset reason: Hardware button
[000632] [INIT] Settings size: 351 bytes
[000632] [INIT] Free heap: 31216 bytes
[000633] [INIT] Power: 3160 mV
[000634] [INIT] Power saving delay value: 10 ms
[000639] 
[000649] [TELNET] Listening on port 23
[000651] [WEBSERVER] Webserver running on port 80
[000654] [RELAY] Retrieving mask: 1
[000654] [RELAY] Relay #0 boot mode 0
[000655] [RELAY] Relay #1 boot mode 0
[000657] [RELAY] #0 set to OFF
[000660] [RELAY] #1 set to OFF
[000663] [RELAY] Number of relays: 2
[000666] [BUTTON] Number of buttons: 2
[000669] [LED] Number of leds: 1
[000672] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000686] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000696] [MAIN] Uptime: 0 seconds
[000697] [MAIN] Free heap: 25048 bytes
[000698] [MAIN] Power: 3160 mV
[000817] [WIFI] Creating access point
[000818] [WIFI] MODE AP --------------------------------------
[000818] [WIFI] SSID  ESPURNA_A12C99
[000818] [WIFI] PASS  fibonacci
[000821] [WIFI] IP    192.168.4.1
[000824] [WIFI] MAC   62:01:94:A1:2C:99
[000827] [WIFI] ----------------------------------------------
[000833] [MDNS] OK
[001662] [RELAY] Saving mask: 0
[060006] [MAIN] System OK
[086678] [BUTTON] Button #0 event 1
[086679] [RELAY] #0 scheduled ON in 0 ms
[086690] [RELAY] #0 set to ON
[087691] [RELAY] Saving mask: 1
[098302] [BUTTON] Button #0 event 5

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000436] 

[000436] [INIT] ESPURNA 1.12.4`

Any hints?

xoseperez, 2018-03-11:
Apparently, the button is floating once pressed and when it returns to a HIGH state it triggers a random event. Long long press event in the T1 is configured to reset the board so that's what's doing. Can you try modifying the configuration of the button like this?

```
#define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
```

You will have to change both buttons for the ITEAD_SONOFF_T1_2CH in `code/espurna/config/hardware.h` file.

dote78, 2018-03-12:
Hi again.

No luck. I downloaded the code for 1.12.4, loaded into arduino 1.8.3, updated libraries and made only your changes to hardware.ino and selecting T1_2CH in arduino.h.

This is the sequence of some boots right after the flashing. No buttons pressed, yet they notify so, and it also threw an exception.

```
[000173] [INIT] ESPURNA 1.12.4
[000173] [INIT] xose.perez@gmail.com
[000173] [INIT] http://tinkerman.cat

[000173] [INIT] CPU chip ID: 0xCA8DAE
[000175] [INIT] CPU frequency: 80 MHz
[000178] [INIT] SDK version: 2.1.0(deb1901)
[000182] [INIT] Core version: 2_4_0
[000185] [INIT] Core revision: 
[000188] 
[000189] [INIT] Flash chip ID: 0x144051
[000192] [INIT] Flash speed: 40000000 Hz
[000196] [INIT] Flash mode: DOUT
[000199] 
[000200] [INIT] Flash sector size:     4096 bytes
[000204] [INIT] Flash size (CHIP):  1048576 bytes
[000208] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000214] [INIT] Firmware size:       516128 bytes /  127 sectors
[000220] [INIT] OTA size:            507904 bytes /  124 sectors
[000225] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000231] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000236] [INIT] Empty space:          16384 bytes /    4 sectors
[000242] 
[000243] [INIT] BUILD_FLAGS: 
[000246] [INIT] BOARD: ITEAD_SONOFF_T1_2CH
[000249] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NOFUSS NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000263] [INIT] Last reset reason: External System
[000267] [INIT] Settings size: 369 bytes
[000271] [INIT] Free heap: 27024 bytes
[000274] [INIT] Power: 3362 mV
[000277] [INIT] Power saving delay value: 10 ms
[000281] 
[000291] [TELNET] Listening on port 23
[000293] [WEBSERVER] Webserver running on port 80
[000296] [RELAY] Retrieving mask: 0
[000297] [RELAY] Relay #0 boot mode 0
[000297] [RELAY] Relay #1 boot mode 0
[000299] [RELAY] #0 set to OFF
[000302] [RELAY] #1 set to OFF
[000305] [RELAY] Number of relays: 2
[000308] [BUTTON] Number of buttons: 2
[000311] [LED] Number of leds: 1
[000314] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000329] [NOFUSS] Disabled
[000330] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000340] [MAIN] Uptime: 0 seconds
[000341] [MAIN] Free heap: 20760 bytes
[000342] [MAIN] Power: 3360 mV
[000455] [WIFI] Creating access point
[000456] [WIFI] MODE AP --------------------------------------
[000456] [WIFI] SSID  ESPURNA_CA8DAE
[000457] [WIFI] PASS  fibonacci
[000459] [WIFI] IP    192.168.4.1
[000462] [WIFI] MAC   DE:4F:22:CA:8D:AE
[000466] [WIFI] ----------------------------------------------
[000472] [MDNS] OK
[001305] [RELAY] Saving mask: 0
[005807] [BUTTON] Button #0 event 1
[005809] [RELAY] #0 scheduled ON in 0 ms
[005819] [RELAY] #0 set to ON
[006820] [RELAY] Saving mask: 1
[016229] [BUTTON] Button #0 event 5
[... 301 lines cut ...]
[000376] [WEBSERVER] Webserver running on port 80
[000379] [RELAY] Retrieving mask: 2
[000380] [RELAY] Relay #0 boot mode 0
[000380] [RELAY] Relay #1 boot mode 0
[000382] [RELAY] #0 set to OFF
[000385] [RELAY] #1 set to OFF
[000388] [RELAY] Number of relays: 2
[000391] [BUTTON] Number of buttons: 2
[000394] [LED] Number of leds: 1
[000397] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000412] [NOFUSS] Disabled
[000413] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000424] [MAIN] Uptime: 0 seconds
[000424] [MAIN] Free heap: 20760 bytes
[000425] [MAIN] Power: 3353 mV
[000539] [WIFI] Creating access point
[000540] [WIFI] MODE AP --------------------------------------
[000541] [WIFI] SSID  ESPURNA_CA8DAE
[000541] [WIFI] PASS  <REMOVED>
[000544] [WIFI] IP    192.168.4.1
[000547] [WIFI] MAC   DE:4F:22:CA:8D:AE
[000550] [WIFI] ----------------------------------------------
[000556] [MDNS] OK
[001387] [RELAY] Saving mask: 0
```

pascal1969, 2018-03-16:
Hi, the problem of the events on the buttons is solved using the pull-up resistors on the relative gpio pins. 
I'm experiencing other problems. I configured espurna to use it with 16 relays and uart mqtt enabled. Initially everything is ok, after a couple of minutes the connection with the local browser is lost. If you try to call the local web page several times the system crashes. Is there a solution?

pascal1969, 2018-03-16:
I also wanted to say that if I configure espurna to work with a 4 channel sonoff, without uart mqtt active, everything works perfectly. I have one device under test for 5 days without crashes.

pascal1969, 2018-03-17:
If in sonoff4 delete all relay and all button and activate uart mqtt, crash is caused by Alexia support, I have decoded the stack frame. If I disactive Alexia support, all work correctly

dote78, 2018-03-17:
> Hi, the problem of the events on the buttons is solved using the pull-up resistors on the relative gpio pins.

@pascal1969 Hi, could you elaborate a bit more? Does that solve the problem described in the first post?

pascal1969, 2018-03-18:
Yes. I noticed these events on the buttons by programming a module ESP8266-07, while on nodemcu there were not. Then I looked at the wiring diagram of the nodemcu and saw the pull-up resistors on the buttons pins. I inserted the pull-up resistors of 10K on the pins of the ESP8266-07 and the problem disappeared.

xoseperez, 2018-03-19:
@pascal1969 Do you mean it still crashes with no buttons or relays and only due to Alexa?
Can you decode the error stack and copy it here?

pascal1969, 2018-03-19:
@xoseperez after 5 minutes that i have activated alexa support, this is the result:

Exception (28):
epc1=0x4021c273 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

ctx: cont 
sp: 3fff2370 end: 3fff25a0 offset: 01a0

>>>stack>>>
3fff2510:  30303853 3b224335 3d766d20 302e3322  
3fff2520:  0a0d3b22 3fff0f44 3fff0f28 4021c35d  
3fff2530:  3ffeab80 9701a8c0 3ffeab80 9701a8c0  
3fff2540:  00008871 3fff2530 00000000 40220c38  
3fff2550:  3fffdad0 3fff0fc0 00000000 3fff1574  
3fff2560:  3fffdad0 3fff0fc0 0000000b 40205bb8  
3fff2570:  3fffdad0 3fff0fc0 0000000b 40203168  
3fff2580:  3fffdad0 00000000 3fff156e 40223fa8  
3fff2590:  feefeffe feefeffe 3fff1580 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

pascal1969, 2018-03-19:
Decoding 6 results
0x4021c35d: fauxmoESP::handle() at C:\Documents and Settings\mio\Documenti\Arduino\libraries\xoseperez-fauxmoesp-50cbcf3087fd\src/fauxmoESP.cpp line 388
0x40220c38: minute() at C:\Documents and Settings\mio\Documenti\Arduino\libraries\Time-master/Time.cpp line 303
0x40205bb8: alexaLoop() at D:\Temp\Arduino\Espurna SONOFF MIO\espurna-master\code\espurna/debug.ino line 184
0x40203168: loop at D:\Temp\Arduino\Espurna SONOFF MIO\espurna-master\code\espurna/debug.ino line 184
0x40223fa8: loop_wrapper at C:\Documents and Settings\mio\Impostazioni locali\Dati applicazioni\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 124
0x4010070c: cont_norm at C:\Documents and Settings\mio\Impostazioni locali\Dati applicazioni\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109

gn0st1c, 2018-03-19:
why are you using 2.4.0-rc2?

pascal1969, 2018-03-19:
@gn0st1c  because I'm testing different devices with integrated esp8266, and in this case I'm testing a mega2560 + esp8266 card. What is the latest firmware stable version for esp8266?

pascal1969, 2018-03-19:
@gn0st1c  I'm sorry, I had missed referring to the esp8266 library. The last one I have is 2.4.1, do you think you can change something?

gn0st1c, 2018-03-19:
[latest stable is 2.4.1](https://github.com/esp8266/Arduino/releases/tag/2.4.1)

pascal1969, 2018-03-19:
ok, with version 2.4.1 it seems to work. I'll keep it under test until tomorrow and then I'll let you know. thank's

pascal1969, 2018-03-19:
Unfortunately, the situation has worsened. You no longer see the crash but the device freezes and there is no way to restart it if you do not remove the power. I'm back to version 2.4.0-rc2, at least with Alexa disabled works without problems.

dote78, 2018-03-22:
`Yes. I noticed these events on the buttons by programming a module ESP8266-07, while on nodemcu there were not. Then I looked at the wiring diagram of the nodemcu and saw the pull-up resistors on the buttons pins. I inserted the pull-up resistors of 10K on the pins of the ESP8266-07 and the problem disappeared.`

Is this something we can achieve by modifying Espurna in any way? Or is it a physical modification?
Or is it a physical modification?

pascal1969, 2018-03-24:
@dote78  it is a physical modification

xoseperez, 2018-03-24:
@dote78 @pascal1969 You could also add BUTTON_SET_PULLUP to BUTTON#_MODE in hardware.ino.

pascal1969, 2018-03-25:
@dote78 I do not know if it is sufficient to enable the pull-up on buttons by software. For example yesterday I lost 2 hours on an esp8266-07 that reset continuously. The problem was that he wanted the pin ch_pd connected to 3.3v, and this despite the pull-up resistor already existed on that pin. I even measured it with the tester to verify that it was good, but the esp8266 did not want to know. 

@xoseperez I have another big problem. I can't use the PZEM004T in any way. Yet with other very few lines of software written by me, it works perfectly, including MQTT support. I need to control a relay according to the values ​​given by the PZEM004T. A simple explanation is possible, since it is not sufficient to enable the relevant option in the arduino.h file #define PZEM004T_SUPPORT       1

dote78, 2018-03-25:
@xoseperez  > You could also add BUTTON_SET_PULLUP to BUTTON#_MODE in hardware.ino.

I did, and did not work. I posted the console output of several of the continuous reboots after enabling that on both buttons. See some comments above for the console output, maybe it will give you a hint of where the problem is. Thanks a lot.

xoseperez, 2018-04-04:
@pascal1969 What changes did you have to do to the code to make PXEM00T work?

pascal1969, 2018-04-05:
@xoseperez unfortunately with espurna I did not find any way to make it work. I tried to import existing working code inside espurna, but it caused a continuous crash, so at the moment I left the tests because I'm dedicating myself to something else.

dote78, 2018-04-12:
@xoseperez Xose, is there any other test you need me to do in order to further diagnose the issue described in my original problem about T1 UE rebooting itself?

xoseperez, 2018-04-12:
I think the problem might be the SoftwareSerial library that is very unstable. I'm experiencing continuous reboots on a V9291F-based power sensor that uses SoftwareSerial.

gn0st1c, 2018-04-13:
@xoseperez : i've read somewhere, not sure where, that using AltSoftSerial solved their issues (with SoftwareSerial)

https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html

gn0st1c, 2018-04-13:
@xoseperez : also there is this but i haven't tried this myself.
https://github.com/SlashDevin/NeoSWSerial

xoseperez, 2018-04-13:
@PaulStoffregen has explicitly reported that AltSoftSerial does not support ESP8266 (https://github.com/PaulStoffregen/AltSoftSerial/issues/44). As for NeoSWSerial, ESP8266 is neither in the supported platforms and I see a lot of AVR specific code in there...

gn0st1c, 2018-04-13:
@xoseperez : i haven't tried neoswserial but i used altsoftserial with esp8266 before. (it was 3 years ago, so maybe older versions was different).

also found this but a new library means testing.. *sigh*
https://github.com/plerup/espsoftwareserial

xoseperez, 2018-04-16:
Well, this is the library I'm already using :)

https://github.com/xoseperez/espurna/blob/68c03202982a84cf456c646eafe066ecc2de4240/code/platformio.ini#L53

dote78, 2018-05-01:
@xoseperez  Hello. I've tried with that library (I think I had it already) with 1.12.5, and disabling support for alexa, influxdb, hass, domoticz, scheduler and some other things that i don't really need as i only use mqtt, but it keeps doing the same. 

Is there anything else you want me to test in order to help you? This is one of the boot cycles and the next reboot triggering event. It keeps doing that with button 0, with me not touching anything in between.

The board looks way different from the one in the wiki, yet the procedure to enter flash mode works. Maybe T1-EU is a totally different hardware from T1-UK?

```
[000167] [INIT] ESPURNA 1.12.5
[000167] [INIT] xose.perez@gmail.com
[000167] [INIT] http://tinkerman.cat

[000167] [INIT] CPU chip ID: 0xA12C99
[000169] [INIT] CPU frequency: 80 MHz
[000172] [INIT] SDK version: 2.1.0(deb1901)
[000176] [INIT] Core version: 2.4.0
[000179] [INIT] Core revision: 
[000182] 
[000183] [INIT] Flash chip ID: 0x144051
[000186] [INIT] Flash speed: 40000000 Hz
[000190] [INIT] Flash mode: DOUT
[000193] 
[000194] [INIT] Flash sector size:     4096 bytes
[000198] [INIT] Flash size (CHIP):  1048576 bytes
[000202] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000208] [INIT] Firmware size:       493600 bytes /  121 sectors
[000214] [INIT] OTA size:            532480 bytes /  130 sectors
[000219] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000225] [INIT] Empty space:          16384 bytes /   4 sectors
[000230] 
[000232] [INIT] BOARD: ITEAD_SONOFF_T1_2CH
[000235] [INIT] SUPPORT: BROKER DEBUG_SERIAL DEBUG_TELNET INFLUXDB MDNS_SERVER MQTT NTP TELNET TERMINAL WEB

[000338] [INIT] Last reset reason: Hardware button
[000339] [INIT] Settings size: 351 bytes
[000340] [INIT] Free heap: 29024 bytes
[000341] [INIT] Power: 3428 mV
[000341] [INIT] Power saving delay value: 10 ms
[000346] 
[000355] [TELNET] Listening on port 23
[000357] [WEBSERVER] Webserver running on port 80
[000360] [RELAY] Retrieving mask: 1
[000361] [RELAY] Relay #0 boot mode 0
[000361] [RELAY] Relay #1 boot mode 0
[000364] [RELAY] #0 set to OFF
[000367] [RELAY] #1 set to OFF
[000370] [RELAY] Number of relays: 2
[000373] [BUTTON] Number of buttons: 2
[000376] [LED] Number of leds: 1
[000379] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000391] [MAIN] Uptime: 0 seconds
[000392] [MAIN] Free heap: 23264 bytes
[000393] [MAIN] Power: 3434 mV
[000509] [WIFI] Creating access point
[000510] [WIFI] MODE AP --------------------------------------
[000511] [WIFI] SSID  ESPURNA_A12C99
[000511] [WIFI] PASS  fibonacci
[000514] [WIFI] IP    192.168.4.1
[000517] [WIFI] MAC   62:01:94:A1:2C:99
[000520] [WIFI] ----------------------------------------------
[000526] [MDNS] OK
[001369] [RELAY] Saving mask: 0
[019566] [BUTTON] Button #1 event 1
[019568] [RELAY] #1 scheduled ON in 0 ms
[019579] [RELAY] #1 set to ON
[020579] [RELAY] Saving mask: 2
[030513] [BUTTON] Button #1 event 5

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
[000162] 
```

xoseperez, 2018-05-08:
Bfff...
I thing we should go back to the start. The button is not floating, right? It either has a pull up resistor or you set the BUTTON_SET_PULLUP flag in the configuration.
Is there something external that could be causing the clicks? Metal?
If you actually touch it, what events does it report?

dote78, 2018-05-20:
Ok, now this is absolutely unexpected (and ridiculous). 

Going back to the start I made two (desperate) tests. First, after flashing (1.12.5 again with most of the features disabled) I tried to boot up the thing without the TX and RX of the USB-Serial connected, just to see if there was any interference on being connected. The same thing happened, continuous "ghost" button clics. Then I mounted the whole thing and booted it up giving the power through the mains, with no USB-Serial connected. Incredibly, this way it works perfectly. No ghost events, no reboots, nothing. I monitored it for five hours yesterday thru the debug screen, and even connected it to my MQTT server and it worked flawlessly. This morning I booted it up the same way, and flashed another one with the prebuilt 1.12.6 binary, and the same thing. They've been up for more than 8 hours now without a glich. Now I have upgraded the original one from 1.12.5 to the prebuilt 1.12.6 and it is working flawless as well.

I'm both angry and happy...

So either powering it from the USB-Serial makes "something" that causes these events, or not being covered and connected to the AC/DC part of the device makes it. It must be noted that the USB-Serial is the same I've been used to flash other 7 sonoffs and wemos things without problems.

Incredible!

xoseperez, 2018-05-21:
Good news! I bet the problem is that the USB-Serial does not provide enough power to those boards, for whatever reason... Anyway I guess we could close this ghost problem :smile: 
Thank you!

closed 2018-05-21 by xoseperez

dote78, 2018-05-21:
Thanks a lot Xose!