xoseperez/espurna #663: 1ch_inching relay always on after boot if pulse mode active

Interpyme, 2018-03-12:
Hi, 

I have a little issue, I´ve tried several 1ch_inching with the same result, when you reboot the device, even if you select any option in switches (boot mode: always off, same as before...) the relay boot in mode on. I have the pulse mode active (0,5s). 

Thanks for you great work!

xoseperez, 2018-03-12:
Sorry, I don't understand the issue. Relay boot mode and pulse mode are different settings... can you please clarify? Thank you.

Interpyme, 2018-03-12:
Sorry, I´ll try to clarify,

Screenshot1:  http://prntscr.com/iq3ln2
Screenshot2:  http://prntscr.com/iq3mzr

I have the 1ch_inching connected to a garage door. I have it configured with pulse mode. When I activate the switch0, it opens, stops o close the door.
If reboot the device, the garage door opens, it activates the switch0.
I have attached two different screenshots, but It does the same with one configuration or the other.

I hope I´ve explained myself a little better, sorry but english is not my main language.

Thanks a lot!!!

quikote, 2018-05-07:
It happens the same with my 1 channel inching module. It doesn’t matter which setup you define for boot state and so on.... everytime the module is rebooted the relay is triggered. Also, it doesn’t matter which option do you use in the mqtt disconnecting behavior. The problem is that this module is used to control a garage door, so each time the relay is triggered cause a reboot (for example due to a drop in the mains) the door is activated.

xoseperez, 2018-05-29:
I have been trying to make it work with the 1CH and it's tricky. Problem is that there is no way to know the real state of the relays. At the time being, I would recommend you to use a different board.

Interpyme, 2018-06-15:
Thanks. Which relay's board will you recommend that works with your firmware for open a garage door?

xoseperez, 2018-06-16:
Pretty much any Sonoff board will do. You don't have to have inching mode in the board itself since ESPurna already has that feature.

Interpyme, 2018-06-16:
Thanks for your support. Are you sure I can use a Sonoff Pow, for example, to open a garage door? I need a relay that can have NC / NO mode and don´t use 220V AC. It has to work with DC. Thanks again!

KBrownConsulting, 2018-08-14:
First, since this is my first comment here, let me start by saying thank you so much for such a great firmware. I appreciate it so much! 

Now, with regard to this issue, I would like to provide another piece of information that I think may be relevant since this issue is affecting me as well. (I am using a 1CH as a garage door opener which obviously means it's important it doesn't trigger the relay at unexpected times!)

First, my symptoms are exactly the same as Interpyme described above in the screenshots he attached: Every time I press the Reboot button, regardless of what I've set the Switch settings to, the relay is activated _momentarily_. 

The key piece of information is that the relay is **_ONLY_** activated when the Reboot is pressed. The relay is _not_ activated when I remove power from the board & then plug it back in. This suggests to me that the relay is not being activated during the boot up process but rather that there is some logic within the Reboot button press itself that is toggling the relay!

My question: At least when the 1CH "Boot mode" option is set to "Same as before" couldn't pressing the Reboot button simply reboot the device without toggling the relay??

xoseperez, 2018-08-14:
Unless the issue is with the SDK reset routine, ESPurna does not manipulate the GPIO when it reboots... The only other thing it does is store the reset reason into EEPROM.

Maybe you can try resetting directly without waiting for the EEPROM commit. Replace this line:

https://github.com/xoseperez/espurna/blob/13b0279ecc23ff811a2118ff12281f15b3c08aed/code/espurna/ws.ino#L149

with:

```
ESP.restart();
```

KBrownConsulting, 2018-08-14:
As shown in this [screenshot](https://prnt.sc/kiiezk), I made the suggested code change & reflashed the device, but unfortunately the same behavior persists: Clicking the Reboot button causes the relay to activate momentarily but unplugging the device & plugging it back in does NOT cause the relay to activate...

Does this imply the problem is indeed in the SDK reset routine? Is that something you can investigate or is that outside your control? Is there anything else I can test to try and help get to the bottom of the issue?

KBrownConsulting, 2020-08-11:
As a follow up 2 years later... While I don't know the status of this bug in ESPurna, I can say that I just tested this in Tasmota v8.4 and the bug no longer seems to exist! (I don't know which version of Tasmota it was fixed in since I jumped from v6.1.1 where the bug did exist, to v8.4 which appears to be working properly.)

I realize this comment is about Tasmota, but I figured I'd at least report that there does appear to be a way to fix this bug, since when it was reported here it was affecting both Tasmota & ESPurna. Again, sorry if this is off-topic, but since there have been no other responses since my previous comment, I thought I'd at least give a status update in case it's helpful.

mcspr, 2020-08-11:
As noted in the comments above, it seems that the issue is because of Arduino Core briefly changing relay pin mode from output to input, turning GPIO state to 1 very briefly.
Tasmota overrides Arduino Core files (see __resetPins() in tasmota/support_cores.ino and tasmota/core_esp8266_wiring_digital.cpp, which is copied from the Core for some reason), disabling this behaviour. In addition, allowing switch to remain ON between device reboots.

You can easily test this exact approach when building the firmware by changing the original __resetPins() in the ESP8266 Arduino Core's [core_esp8266_wiring_digital.cpp](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring_digital.cpp) in the folder where the Core files are installed.