xoseperez/espurna #664: DST by region

WayneManion, 2018-03-12:
This is a pretty first-world kind of problem, but Espurna seems to follow European rules for Daylight Savings Time. All my devices think it's 7:17 right now, but in America it is 8:17 because the US does its Spring clock adjustment the second Sunday in March and Europe does it on the last Sunday in March. 

The US Fall adjustment happens the first Sunday in November vs. the last Sunday in October in Europe. 

Is there a simple way to implement regional DST-following in Espurna? Perhaps default to Euro rules and have a checkbox to use the US dates?

sashimanu-san, 2018-03-13:
Then the rest of the world will complain too, hihi

There's [tzdata](https://en.wikipedia.org/wiki/Tz_database) database for that.

xoseperez, 2018-03-13:
True, but let's walk step by step. I have done a PR against the NTP library ESPurna uses to include DST calculation for different zones and implemented USA calculations.

https://github.com/gmag11/NtpClient/pull/58

xoseperez, 2018-03-24:
Staged for release. At the moment it requires NTPClientLib develop branch.

closed 2018-03-24 by xoseperez