xoseperez/espurna #667: Blank page on iOS Safari

sashimanu-san, 2018-03-13:
In stock Safari on an iPhone, the Espurna control page renders totally blank, making the device uncontrollable from that device.

User agent is `Mozilla/5.0 (iPhone; CPU iPhone OS 11_2_6 like Mac OS X) AppleWebKit/604.5.6 (KHTML, like Gecko) Version/11.0 Mobile/15D100 Safari/604.1`

[image]

gn0st1c, 2018-03-13:
probably because safari has strict rules when it comes to headers. content-type, length etc.

thebradleysanders, 2018-03-14:
It seems to be an error with the authentication in Safari.

sashimanu-san, 2018-03-14:
Can't agree.

The auth prompt is displayed and, if the credentials are correct, not displayed again.

xoseperez, 2018-03-15:
It's probably an error in websocket auth. It might be weird but you can log in using a different browser and disable websocket authentication in the web UI, admin tab. The you should be able to use Safari again. Check d9cd999db9ac57a56a59fe927901b6a84f8710b2.

zerog2k, 2018-03-16:
is there a way to disable WS Auth during fw build time, maybe macro or something?
Is it adding this as compiler define?
`-DNOWSAUTH` ?

xoseperez, 2018-03-19:
Set `WS_AUTHENTICATION` to 0 in `general.h`.

TarjeiD, 2018-03-24:
The compatibility with Safari broke with [694d70d](https://github.com/xoseperez/espurna/commit/694d70d0974f93cc77103ed9cd8470d671061075) where ESPAsyncWebServer was updated to #232b87a.
Going back to using #313f337, will also fix the issue with Safari on both MacOS and iOS.

xoseperez, 2018-03-24:
Going back to #313f337 means disabling WS authentication for all users. 694d70d was meant to enable them while allowing users having problems with Safari to disable it manually, either via a compile setting or in the web UI. Of course, you should use a different browser to log in the web UI and disable WS authentication, once done you can use Safari again.

zerog2k, 2018-03-24:
Not so familiar with the wsauth option, but wondering if it would be possible to detect "problematic" browser user-agent to dynamically switch to different auth for the session? Is this something implemented in espurna, or in the ESPAsyncWebServer codebase?

xoseperez, 2018-03-24:
Not in ESPurna, right now it's a manual setting. I don't know how such a feature could be implemented. User agent maybe?

TarjeiD, 2018-03-26:
I like your focus on security. That's why I settled on ESPurna over ESPEasy and Sonoff-Tasmota.
Why can't the authentication be performed using http before switching to WS?

chryss44, 2018-03-31:
Hi all,
I have tried  on Safari MacBook (Sierra) with Firmware v1.12.1 and work well.
On 1.12.4 not work and blank page with this message on web dev console : "WebSocket connection to 'ws://192.168.44.65/ws' failed: Unexpected response code: 401 "

xoseperez, 2018-04-03:
@chryss44 There was a change in 1.12.4 to recover ws auth for security reasons and add the option to disable it manually. Websocket communication in 1.12.1 is open.

@TarjeiD Yes, that was the original approach and I changed it when ESPAsyncWebServer recovered the ws auth method. I guess it was a bad move.

l3d00m, 2018-04-25:
I am having this problem with firefox as well, but only if I connect from a different network (/subnet). If I'm in the network (192.168.110.0/24) it does render without problems but If I'm in a different network (eg 192.168.100.0/24) that has a route to the first network, it just displays the blank pange and "ESPurna 0.0.0" as the title. 

Console output

```
Firefox can’t establish a connection to the server at ws://192.168.110.51/ws. 192.168.110.51:3:18172
```

skyynet, 2018-05-11:
I wanted to use an old iPod Touch (4. Gen) with iOS 6.1.6 to switch my Electrodragon WiFi relay board and also get a blank page in Safari. On my iPhone X with iOS 11.3.1 it works just fine. WS Auth is turned off.

xoseperez, 2018-05-13:
Sorry, I hope you can understand it's impossible to support any possible hardware out there. I can only test the web interface on Chrome, FF for Linux and Android.

skyynet, 2018-05-13:
Of course I see the point. I thought you might have experience with the behavior of older Mobile Safari versions which could workaround the issue. Since I just programmed a WebApp which calls the relay board via REST API it's not a problem for me.

0anton, 2018-06-09:
Hi,

I am using ESPURNA 1.12.6 on Sonoff Pow switch. Web-Interface does not load (blank page) on both Safari and Chrome browsers on Iphone and Ipad. Authentication pop-up shows up and after entering the password, the black page is rendered. Switching off Web-Socket authentication (WS Auth) in the web-console does not help.

Any ideas how to make web-page to show on Apple devices?

Kind regards,
Anton

mcspr, 2018-06-09:
@0anton dev branch started using different authentication method for websockets, can you try https://github.com/xoseperez/espurna/commit/dff81d6795c4b6647463645c2f484be578d10562 build?

xoseperez, 2018-06-11:
@0anton @skyynet @chryss44 @TarjeiD @sashimanu-san Can you test dev branch?

0anton, 2018-06-19:
Hi Xose,

I could test the dev branch if upgrade is possible over web form. Is it the
case or I need a to attach a programmer?

Kind regards,
Anton Golubev

xoseperez, 2018-06-22:
Released with 1.13.0. Closing.

closed 2018-06-22 by xoseperez

zerog2k, 2018-06-30:
Working much better with 1.13.0 thanks @xoseperez 
One comment/feedback is that while I'm able to log into the web ui on IOS Safari, I get sent to the change password page. I try to change the password, and it doesn't seem to do anything. However, I tried to login on another device (ipad), and the password seems like it was changed. So something about the initial change password prompt. (In general, it would be nice to be able to defer the password change (like for 1 time, or 24 hours, etc), so I can at least login to the device, and join it to my wifi network, then switch over to my PC for full configuration, where I would be happy to change password (and can use password manager ;)

SirJMD, 2018-07-05:
I just flashed my Sonoff RF bridge with 1.13.0, but I can't load the web gui from my computer using chrome. From my phone it works fine.

thebradleysanders, 2018-07-09:
Do you get the Web Auth prompt in Chrome?

SirJMD, 2018-07-09:
Nope, nothing. Blank page only.