xoseperez/espurna #669: Wall switch with Wemos D1 mini

juliusedward, 2018-03-13:
Hi Xose,

I am a big fan of your work and use espurna to control Wemos D1 mini boards in my home connected to few lights and plugs.

I am trying to make the Wemos D1 mini work with a single channel relay connected to D1 pin and a wall switch connected between D3 and GND.

This is my hardware.h

#elif defined(WEMOS_D1_MINI_RELAYSHIELD)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "D1_MINI_RELAYSHIELD"

    // Buttons
    // No buttons on the D1 MINI alone, but defining it without adding a button doen't create problems
    #define BUTTON1_PIN         0   // Connect a pushbutton between D3 and GND,
                                    // it's the same as using a Wemos one button shield
    #define BUTTON1_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          5
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // When Wemos relay shield is connected GPIO5 (D1) is used for relay,
    // so I2C must be remapped to other pins
    #define I2C_SDA_PIN         12  // D6
    #define I2C_SCL_PIN         14  // D5

The problem now is if the wall switch is in ON position and the board is reset, it never boots up. I have to change the switch to off position to boot it up again

I see this weird line in serial monitor
https://ibb.co/gBcdec

Also if the switch is in off position and the deice is reset, this same line repeats 3 times before the 
https://ibb.co/fFTqsx

Complete log with switch in off position
[000085] [INIT] ESPURNA 1.12.5a
[000085] [INIT] xose.perez@gmail.com
[000085] [INIT] http://tinkerman.cat

[000086] [INIT] CPU chip ID: 0x8DCC77
[000087] [INIT] CPU frequency: 80 MHz
[000091] [INIT] SDK version: 2.2.1(cfd48f3)
[000094] [INIT] Core version: 2_4_1
[000098] [INIT] Core revision: 
[000100] 
[000101] [INIT] Flash chip ID: 0x164020
[000105] [INIT] Flash speed: 40000000 Hz
[000108] [INIT] Flash mode: DIO
[000111] 
[000112] [INIT] Flash sector size:     4096 bytes
[000117] [INIT] Flash size (CHIP):  4194304 bytes
[000121] [INIT] Flash size (SDK):   4194304 bytes / 1024 sectors
[000126] [INIT] Firmware size:       498368 bytes /  122 sectors
[000132] [INIT] OTA size:            548864 bytes /  134 sectors
[000138] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000143] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000149] [INIT] Empty space:          16384 bytes /    4 sectors
[000154] 
[000155] [INIT] BUILD_FLAGS: 
[000158] [INIT] BOARD: WEMOS_D1_MINI_RELAYSHIELD
[000162] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000175] [INIT] Last reset reason: External System
[000179] [INIT] Settings size: 1104 bytes
[000183] [INIT] Free heap: 31376 bytes
[000187] [INIT] Power: 3002 mV
[000189] [INIT] Power saving delay value: 10 ms
[000193] 
[000208] [TELNET] Listening on port 23
[000209] [WEBSERVER] Webserver running on port 80
[000213] [RELAY] Retrieving mask: 0
[000213] [RELAY] Relay #0 boot mode 0
[000213] [RELAY] #0 set to OFF
[000215] [RELAY] Number of relays: 1
[000217] [BUTTON] Number of buttons: 1
[000221] [LED] Number of leds: 1
[000223] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000239] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000260] [MAIN] Uptime: 0 seconds
[000260] [MAIN] Free heap: 25752 bytes
[000261] [MAIN] Power: 3005 mV
[000353] [BUTTON] Button #0 event 2
[000354] [RELAY] #0 scheduled ON in 0 ms
[000420] [RELAY] #0 set to ON
[000433] [WIFI] Scanning
[001422] [RELAY] Saving mask: 1
[002632] [WIFI]     BSSID: 80:26:89:02:FC:E5 CH:  1 RSSI: -70 SEC: AUTO SSID: Bruno2.4
[002633] [WIFI] --> BSSID: 30:B5:C2:AF:F0:54 CH:  1 RSSI: -42 SEC: AUTO SSID: Julius_Extended
[002637] [WIFI]     BSSID: C8:B3:73:16:77:A0 CH:  1 RSSI: -71 SEC: WPA2 SSID: Julius
[002645] [WIFI]     BSSID: 0C:D2:B5:72:A1:4C CH:  5 RSSI: -77 SEC: WPA  SSID: Nitesh
[002652] [WIFI]     BSSID: D0:5B:A8:D2:0D:32 CH:  7 RSSI: -83 SEC: WPA2 SSID: airtel_D20D32
[002660] [WIFI]     BSSID: 74:DA:DA:EB:9C:C9 CH:  9 RSSI: -70 SEC: AUTO SSID: PU
[002667] [WIFI]     BSSID: BC:8A:E8:0B:FA:6C CH: 11 RSSI: -80 SEC: WPA2 SSID: JioFi2_0BFA6C
[002675] [WIFI] Connecting to BSSID: 30:B5:C2:AF:F0:54 CH: 01, RSSI: -42, SEC: AUTO, SSID: Julius_Extended
[042815] [WIFI] MODE STA -------------------------------------
[042815] [WIFI] SSID  Julius_Extended
[042816] [WIFI] IP    192.168.1.148
[042816] [WIFI] MAC   EC:FA:BC:8D:CC:77
[042819] [WIFI] GW    192.168.1.1
[042822] [WIFI] DNS   8.8.4.4
[042825] [WIFI] MASK  255.255.255.0
[042828] [WIFI] HOST  ESPURNA_8DCC77
[042831] [WIFI] BSSID 30:B5:C2:AF:F0:54
[042835] [WIFI] CH    1
[042837] [WIFI] RSSI  -55
[042839] [WIFI] ----------------------------------------------
[042845] [MDNS] OK
[043847] [MQTT] MQTT brokers found: 0
[046430] [NTP] Error: NTP server not reachable
[046517] [NTP] Time: 2018-03-13 15:51:34
[049280] [WEBSERVER] Request: GET /index.html
[049858] [WEBSOCKET] #1 connected, ip: 192.168.1.119, url: /ws
[051913] [WEBSOCKET] Requested action: relay
[051915] [RELAY] #0 scheduled OFF in 0 ms
[051917] [RELAY] #0 set to OFF
[052918] [RELAY] Saving mask: 0
[060010] [MAIN] System OK

I do not know what I am doing wrong. Please help me with this

thebradleysanders, 2018-03-14:
If GPIO 0 is shorted to GND during boot up, the device will go into program mode, to fix this attach your switch to a different GPIO pin, by default this function was meant for a momentary switch such as a button.
Don't forget to set the new GPIO Pin in your hardware.h file.

juliusedward, 2018-03-15:
Thanks. I used D1 and D2 and it works great. Thank you
If I want to use a 4 channel relay with 4 wall switches, which pins can I use?

THis issue can be closed anyway. Thnaks for the support

closed 2018-03-15 by juliusedward

rossmax, 2018-08-13:
Can you share you code? How you control led strip with espurna using wemos d1 mini?