xoseperez/espurna #670: Bulbs Freezing

jonny190, 2018-03-13:
i have 2 Ai thinker bulbs, 2 magic home rgb and a sonoff touch all running 1.12.4 that seem to freeze after a few minuates and need to be restarted to regain wifi and in the touch's case ability to switch, any ideas?

jonny190, 2018-03-13:
Just for detail this was only when compiling my own firmware as most of my RGB strips are GRB format when using magic home plus its handy to pre-program config

thebradleysanders, 2018-03-14:
Please Visit the Releases Page: https://github.com/xoseperez/espurna/releases/
find your device and apply the image to your device with the OTA upload in the Web UI.

There may have been an issue with your self-compiled firmware.

xoseperez, 2018-03-15:
This could be due to using lwIP 2.0 (unstable). Check #572.
In the [Arduino IDE wiki page](https://github.com/xoseperez/espurna/wiki/ArduinoIDE) it is recommended to use IwIP 1.4 due to stability issues with the latest version.

jonny190, 2018-03-16:
@thebradleysanders Thanks for you reply, the pre-compiled images seem to be working however i need to compile my own to swap the colors of pins over.

@xoseperez thanks for the heads up i'm compiling my own now and i'll let everyone know how it works after testing

jonny190, 2018-06-14:
This is corrected issues

closed 2018-06-14 by jonny190