xoseperez/espurna #674: Error: Detected unknown package 'tool-scons'

Portisch, 2018-03-14:
root@ubuntu:/opt/espurna/code# ./build.sh
--------------------------------------------------------------
ESPURNA FIRMWARE BUILDER
Building for version 1.12.5a
--------------------------------------------------------------
Building web interface...
[15:50:34] Using gulpfile /opt/espurna/code/gulpfile.js
[15:50:34] Starting 'buildfs_inline'...
[15:50:36] Finished 'buildfs_inline' after 2.54 s
[15:50:36] Starting 'buildfs_embeded'...
[15:50:36] Finished 'buildfs_embeded' after 123 ms
[15:50:36] Starting 'default'...
[15:50:36] Finished 'default' after 3.47 μs
--------------------------------------------------------------
Building firmware images...
* espurna-1.12.5a-aithinker-ai-light.bin
CorePackageManager: Installing tool-scons @ ~2.20501.4
Error: Detected unknown package 'tool-scons'

I tried on windows and ubuntu.
How to fix this?

xoseperez, 2018-03-14:
Have you tried building a single environment using PlatformIO first?

The build script takes care of building the PlatformIO environments during the testing process in Travis and also the deployment images, but it does not install the build environment.

xoseperez, 2018-04-03:
@Portisch have you managed to build the project?

Portisch, 2018-04-03:
I didn't invest a lot of time in this right now. I got it compiled on Ubuntu but not on windows.

closed 2018-04-03 by Portisch