xoseperez/espurna #675: continuous reset on arilux E27

gnkarn, 2018-03-14:
i have two equal lights , both burned with the same firmware , only change is of course IP , and serverName.

on one of them , the esp is reseting , with this message

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000078] [MAIN] System UNSTABLE
[000136]

[000136] [INIT] ESPURNA 1.12.3
[000136] [INIT] xose.perez@gmail.com

[000136] [INIT] http://tinkerman.cat

[000136] [INIT] CPU chip ID: 0xAF54B1
[000138] [INIT] CPU frequency: 80 MHz
[000142] [INIT] SDK version: 1.5.3(aec24ac9)
[000145] [INIT] Core version: 2_3_0
[000149] [INIT] Core revision:
[000151]
[000153] [INIT] Flash chip ID: 0x1440C8
[000156] [INIT] Flash speed: 40000000 Hz
[000159] [INIT] Flash mode: DOUT
[000162]
[000163] [INIT] Flash sector size:     4096 bytes
[000168] [INIT] Flash size (CHIP):  1048576 bytes
[000172] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000177] [INIT] Firmware size:       479552 bytes /  118 sectors
[000183] [INIT] OTA size:            544768 bytes /  133 sectors
[000189] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000194] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000200] [INIT] Empty space:          16384 bytes /    4 sectors
[000205]
[000207] [INIT] BOARD: ARILUX_E27
[000209] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER TELNET TERMINAL WEB

[000221] [INIT] Last reset reason: Hardware Watchdog
[000226] [INIT] Settings size: 1043 bytes
[000230] [INIT] Free heap: 29280 bytes
[000233] [INIT] Power saving delay value: 10 ms

i would like to know if is there something on the configuration , that i can change , to make it work Normal without reseting .
thanks

gnkarn, 2018-03-15:
i made some modifications on the config, and it is working fine now

gnkarn, 2018-03-15:
mo , after a while it continues with random  resets 

[060004] [MAIN] System OK

 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000135]

gnkarn, 2018-03-15:
will try adding 
ESP.wdtFeed()

on the main loop , don't know if this is pertinent though ...

gnkarn, 2018-03-15:
no changes , still same problem

ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000134]

gnkarn, 2018-03-15:
things tested , add yield() on main loop , fail
disable telnet : fail
disable terminal: worse , it added a new error
rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

gnkarn, 2018-03-15:
pusub, instead of async:  more errors

Exception (3):
epc1=0x40225436 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40013c51 depc=0x00000000

ctx: cont
sp: 3fff11c0 end: 3fff1480 offset: 01a0

>>>stack>>>
3fff1360:  00000000 3fff13fc 3fff13d0 4020e9a1
3fff1370:  3fff1390 3fff01a4 00000000 3fff0454
3fff1380:  00000000 00000000 0ccccccc 00000009
3fff1390:  00000019 00000001 00000000 3fff0454
3fff13a0:  3fffdad0 0000002d 00000000 402254ec
3fff13b0:  3fff1578 0000017f 0000017f 40213524
3fff13c0:  3fffdad0 0000002d 3fff13fc 4022c96d
3fff13d0:  4020d64c 00000000 00000000 4020e750
3fff13e0:  3fffdad0 0000002d 00000000 40217c74
3fff13f0:  00000000 00000000 00000000 00000000
3fff1400:  00000000 00000000 3ffefddc 40216fbb
3fff1410:  3fffdad0 3ffefef0 3ffefde4 4021f12b
3fff1420:  3fffdad0 3fff00e8 00000000 4020b76c
3fff1430:  3fffdad0 3fff00e8 3ffefde8 40216e2d
3fff1440:  3fffdad0 3ffefef0 3ffefde4 4021f29d
3fff1450:  00000000 3fff00e8 00000001 40214cd8
3fff1460:  3fffdad0 00000000 3fff044d 4020f504
3fff1470:  feefeffe feefeffe 3fff0460 40211318
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16
tail 8

gnkarn, 2018-03-15:
disable web support :  still errors

 ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000134]

gnkarn, 2018-03-15:
disable home assistant support:  same problem, as with all other , it connects publishes mqqt, and then ....

ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d

csum 0x2d
v09826c6d
~ld
[000135]

gnkarn, 2018-03-15:
my conclusion , as no matter what i disable, the problem is the same , it has to be a hardware issue , 
so i bought 2 lamps , one is good the other is not ,.
quality rate = 50% . too poor to buy more .
if you think otherwise , i would appreciate letting me know .

gnkarn, 2018-03-15:
i save it first , in espurna core mode , and is not reseting , so i again suspect that something on the configuration is causing resets ...

gnkarn, 2018-03-15:
FINALLY  .. , IT SEAMS TO BE WORKING NOW
for some reason it is important to save n CORE mode first, and then use the complete version.

gnkarn, 2018-03-15:
i found a VERY BAD behavior of PLATFORMIO, that could be part or all the cause ...

there are times when you change a file in platfomio , Build and upload, BUT , the version uploaded HAS  NOT CHANGED !!! , it is even a nightmare to change the device IP address , as it keeps , the old IP after upload ...  (??¿)

closed 2018-03-15 by gnkarn

xoseperez, 2018-03-15:
You were probably using latest 2.4.X Arduino Core for ESP8266 and doing OTA upgrades, right? The binary size of an ESPurna image for light bulbs with the latest core is >512Kb which means it does not fit into the available room for OTA. You were probably not upgrading it at all...

The way to upgrade a device via OTA when the binary is too large is doing a [two-step update](https://github.com/xoseperez/espurna/wiki/TwoStepUpdates) flashing first the core version (which is relatively small) and the the full version again.

gnkarn, 2018-04-23:
COMMENT:  xose , just for the record about potential reliability issues with the lights .
I worked with two Arilux e27 , one worked fine with no problems from the beginning , the other was very problematic to get flashed, and once flashed, the connection was unreliable , until it was most of the time disconnected from the network.
i did a final effort , to flash it fresh , from the beginning , but even when succeeded to reflash , it was totally useless .

Before giving up with aril, will buy two more and see , what happens....

earth08, 2019-03-03:
unable to get it working
uploading is done