xoseperez/espurna #676: SONOFF POW cant calibrate power settings Again :(

gibbo520, 2018-03-14:
https://github.com/xoseperez/espurna/issues/194

Same problem in Version 1.12.4

xoseperez, 2018-03-15:
Do you mean the calibration values are not persisted?

gibbo520, 2018-03-15:
Yes !! Again :(

zerog2k, 2018-03-17:
I have KMC-70011 (w/ HLW8012), ESPURNA 1.12.5a
Also, unable to calibrate energy monitor settings - they are way off.
Calibration using 50W resistive incandescent light bulb, and calibration settings are not saving/taking effect.

gn0st1c, 2018-03-17:
@xoseperez : btw, for KMC 70011 HLW8012_VOLTAGE_R_UP is 2 * 1000000 (2 x 105 resistors)

xoseperez, 2018-03-24:
Staged for release

closed 2018-03-24 by xoseperez

karmacoma92, 2018-07-19:
Hi again, something similar is happening in 1.13.1. I just got the POW and it is the first time I use it so I'm not sure if there is any special procedure for the calibration rather than connect it to a bulb, switch it on and write down the vaules under sensors...

karmacoma92, 2018-07-19:
it looks to me the chip at the back is a CSE7759 and not a CSE7766, could that be the problem?
I have tried with 1.12.6 pow and powr2, 1.13.0 POW and POWR2 and 1.13.1 POW and POWR2, none of them allows me to store the values....