xoseperez/espurna #678: delay for Relay and Button

testeve, 2018-03-15:
Hello!

i use the Esp8266 with espurna Sonoff 4CH Firmware  for light in my home.

i switch with a Key and Alexa, its run perfect.

My wish would be that the lamp switches itself off after 3 hours.
Unfortunately, I can only set delay at 3600sec.

Thanks for help!

xoseperez, 2018-03-15:
The pulse functionality uses the Ticker library that internally stores the data in a 32bits variable representing microseconds, that set the maximum value in around 71 minutes, that is why in ESPurna the max pulse time is limited to 60 minutes.

Changing this is possible but not trivial.

mcspr, 2018-06-06:
@xoseperez Pulse timers used (until aaab65f8772ebe9011b7d6356e2ecd1e433ef606) `Ticker.once_ms` which uses sdk `os_timer_arm` that has milliseconds for an argument, so there is much larger limit.

[That is for SDK 2.2.1](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf), `os_timer_arm_us` is the microseconds one

ristomatti, 2018-11-21:
@xoseperez Could the solution be time based? Switch on/off would add a one time schedule to switch off/on when the given amount of time had passed. I'd find this feature really useful.

My use case is a metal halide lamp I use as a winter time "wake up" light. It uses 320W and also heats quite a lot so I need to have an automatic toggling it off. Sometimes I'd like to have it on for longer than an hour but I cannot just switch it back on as the metal halide lamp chemistry prevents it from turning on again before it has cooled down to a certain level. This takes around 20-30 minutes.

taiko000, 2019-08-15:
I'm also interested in a pulse time longer than 60 minutes, for use cases such as charging battery powered devices for only a few hours and lamps turning off after a few hours. 
Wouldn't it be possible to use `<Ticker.once_ms>` to simply call  `<Ticker.once_ms>` again after the maximum time has elapsed for longer periods of time?

mcspr, 2019-08-15:
It would be. One approach that I was thinking about is to use internal counter to measure how long timer was running so far. Something like this is already implemented in Sming Core for usec timer:
 https://github.com/SmingHub/Sming/blob/develop/Sming/Core/Timer.cpp (see `longIntervalCounterLimit`)
Something similar can be implemented here. Instead of loop check, https://github.com/xoseperez/espurna/commit/aaab65f8772ebe9011b7d6356e2ecd1e433ef606, millis value can be attached to the timer and it can periodically compare it to the target time, adjusting the next run either with max value or some smaller one

closed 2022-01-15 by mcspr