xoseperez/espurna #680: Configuring magichome-led-controller-20 with alternative LED strip ordering.

oxcxsxixbxaxn, 2018-03-15:
Hi,
Just switched from tasmota firmware because of the better control of separate RGB and white here in ESPurna but just have one config query.

Is there a separate mapping for the the LED strip wiring?  My LED strip is all wired +GRBW, so although the pinout of the color channels on my magichome is using the correct GPIOs, in my case Red and Green are switched.

[image]

So do I just define a custom magichome hardware.h profile? Or is there some place else I can configure the strip specific ordering?

Cheers...

lobradov, 2018-03-15:
Better location to configure your custom hardware is in `custom.h` which will not be overwritten during `git pull`. With it, you can configure pretty much anything, including mapping of different PWM channels to different GPIO: 

```
    #define LIGHT_CH1_PIN       14      // RED
    #define LIGHT_CH2_PIN       5       // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
    #define LIGHT_CH4_PIN       13      // WHITE

```

oxcxsxixbxaxn, 2018-03-16:
Is custom.h included first?  Will that mean I have to define a whole customer hardware profile in there? or will it override the settings in hardware.h??
```

    #define LIGHT_CH2_PIN       5       // RED
    #define LIGHT_CH1_PIN       12      // GREEN
    #define LIGHT_CH3_PIN       13      // BLUE
    #define LIGHT_CH4_PIN       15      // WHITE

```

rubenzori86, 2018-12-04:
Hi, is possible to change gpio channel order under firmware already installed?

xoseperez, 2018-12-12:
Not yet. I will be with v2 release.

JumpmanJunior, 2018-12-30:
I also would like to switch the channels. I tried commands like the following one:
SET chGPIO0 12
The new configuration is saved but it has no effect. If I understand your comment correctly this will not work at the moment.
So I build a new firmware with my desired settings, performed a firmware upgrade and factory reset, but the new swapped colours are still not working. Is it possible that the GPIO settings can only be changed by flashing a new firmware? Or am I doing something wrong and the firmware upgrade should work?