xoseperez/espurna #684: Accessing from cloud

varungujjar, 2018-03-15:
Hey @xoseperez is there a way I can access & control the switch from cloud using something aRest  apart from MQTT

ESP8266 <-> Cloud Server ( Hosting) -> HTML / PHP (JSON)

lobradov, 2018-03-15:
Would this help?

https://github.com/xoseperez/espurna/wiki/RESTAPI

varungujjar, 2018-03-15:
tried that but i guess that only allows you to access using local ip .. how do I access it from the internet ( like access it from  my office.)  like a websocket to remote server and then using a webpage switch it on & off ? am really not got at describing things :) . but i'll try my best to explain  :)

1) My internet provider does not give me access to a WAN IP
2) so does my personal shared hosting doesnt allow you to host MQTT server
2) Although I can host PHP pages
3) something where ESP POSTS payload to PHP page  & a web socket that waits for GET payload to trigger the device

xoseperez, 2018-03-15:
A few months ago I wrote an article about this topic, how to access you local devices from the Internet in a secure way. You will need some kind of local server, like a Raspberry Pi.

http://tinkerman.cat/secure-remote-access-to-your-iot-devices/

varungujjar, 2018-03-15:
@xoseperez  thank you xose pretty much helps.. :) still the setup is quite huge for a single switch.. however I was also thinking what if  we could POST everytime a change occurs to a php server and use GET (every second) to poll for a change online using https .. and  I can create a php script and MYSQL database that will store the state of the switches..  do you think thats safe  ?

xoseperez, 2018-03-15:
@varungujjar It might be a huge setup but it makes sense on the long run. It frees the ESP8266 in your local network from doing heavy lifting using SSL themselves. POSTing to a known service might be easy to do, polling for new actions every second using GET might be a little overkill.

xoseperez, 2018-07-02:
Closing this since there has been no feedback. Please feel free to reopen if necessary.

Also, check #937 as it could help to configure direct access from outside your network over HTTPS.

closed 2018-07-02 by xoseperez