xoseperez/espurna #686: SONOFF vailability issues

akasma74, 2018-03-16:
Hi All,

I have SONOFF RF bridge (with SONOFF PIR2 and DW1 sensors paired) and SONOFF SV flashed with Espurna 1.12.3. I use Home Assistant 0.64.3 and my sensors configured as MQTT switches listening to state and availability topics.

When I look in my HA logs from tonight, I can see many times when each of them independently became unavailable (meaning it published status 0 message) and then instantly back (SONOFF SV: 4.26AM, 4.43AM, 4.45AM, 5.22AM, 5.51M, 6.08AM etc).

I can easily open RF bridge's web interface, but the SV's one takes longer to open, and sometimes it's a blank page OR Cannot connect to the server (but it works!). Once I had to physically disconnect it from power source (9V power adaptor).

What's going on and how to debug it?

xoseperez, 2018-03-19:
First you should try to find out if the problem is a reboot or a connection issue, you can just look at the uptime of the devices (either on the web UI or MQTT message). There is a report on connectivity issues with ESPurna here #540. 

The `home/sonoff_rfbridge/button/0 1` means the button has been pressed. Check the button event codes here: https://github.com/xoseperez/espurna/wiki/Buttons-and-switches

akasma74, 2018-03-19:
Well, it did not bother me too much (and actually I did not see any more unavailable messages in log file) until yesterday when the SONOFF SV became unavailable on my HA.
It turned out it was on, but blinking rapidly. Disconnecting it from power and powering up again didn't work. I was able to access its web interface, but half to the fields were empty and it acted weirdly - sometimes did not confirm when I pressed Save button, and did not open Select File dialog when I tried to backup its configuration from Admin tab. And if I rebooted it, it cleared all fields to default (but kept its name and WiFi network settings).
I ended up flashing both rf bridge and SV with the current Espurna.
Will keep an eye on them and update.

akasma74, 2018-03-19:
It's still playing up :(
The relay was on, but became unavailable after about 3 hours. LED not blinking at all.
I powered it off and back on, it found WiFi network.. and became unavailable straight on.
The board's components are very hot, I suspect it's overheating, but I don't know why.
I connected it [as described](https://www.itead.cc/smart-home/sonoff-sv.html) on Itead website - default mode, 9V in, no jumper, 9V out.
Any ideas what's wrong?

akasma74, 2018-03-21:
I'll close the issue as the board is dead and I cannot investigate it further :\

closed 2018-03-21 by akasma74