xoseperez/espurna #687: ESP32 porting

zafrirron, 2018-03-17:
Hi,
Great Job!
Any idea what it would take to port this project to ESP32 chip?
thanks

xoseperez, 2018-03-19:
Time. 
Now seriously, it's something I want to do, but it's not a priority.

Sanfam, 2018-03-30:
I can imagine this would allow you to do wonders for Web UI responsiveness, assuming multi-threading across cores was implemented and application/wifi functionality were evenly assigned to each of the two cores.

henkiejan1, 2018-08-04:
I have just a ESP32 development board from Aliexpress from [here](https://nl.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Core-ESP-32-ESP-32S/32802431728.html).  It's under the 5 euro and has bluetooth and 160MHz dualcore processor. Looks ideal for a RF bridge (with a shield) and bluetooth. I stay tuned here :)

skarcha, 2018-10-15:
Hi!

I'm working on this, but I need some help. For example with EEPROM_Rotate library. I'm using EEPROM32_Rotate from @xoseperez , but the API has changed and I'm not sure what should I do.

It's my first project bigger than Blink example but I'm ready to expend time on it. Could you give me some help, @xoseperez. Thanks!

This is the progress right now:
```
Processing wemos-d1mini32 (framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32doit-devkit-v1)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 (Stage) > DOIT ESP32 DEVKIT V1
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Converting espurna.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 64 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.2
|-- <EEPROM> 1.0
|-- <EEPROM32_Rotate> 0.9.0 #c9983cf
|-- <SPI> 1.0
|-- <WiFi> 1.2.7
|-- <Update> 1.0
|-- <ArduinoOTA> 1.0
|-- <Ticker> 3.1.4
|-- <DNSServer> 1.1.0
|-- <AsyncTCP> 1.0.1 #6fab772
|-- <Embedis> 1.2.0 #554f5b5
|-- <JustWifi> 2.0.0 #21a7662
Compiling .pioenvs/wemos-d1mini32/src/espurna.ino.cpp.o
Compiling .pioenvs/wemos-d1mini32/src/fs_math.c.o
Generating partitions .pioenvs/wemos-d1mini32/partitions.bin
Archiving .pioenvs/wemos-d1mini32/libf14/libArduinoJson_ID64.a
Indexing .pioenvs/wemos-d1mini32/libf14/libArduinoJson_ID64.a
Compiling .pioenvs/wemos-d1mini32/libf7a/EEPROM/EEPROM.cpp.o
Compiling .pioenvs/wemos-d1mini32/libd89/EEPROM32_Rotate/EEPROM32_Rotate.cpp.o
Compiling .pioenvs/wemos-d1mini32/lib38e/SPI/SPI.cpp.o
Compiling .pioenvs/wemos-d1mini32/lib658/WiFi_ID870/WiFi.cpp.o
Archiving .pioenvs/wemos-d1mini32/lib38e/libSPI.a
Indexing .pioenvs/wemos-d1mini32/lib38e/libSPI.a
Compiling .pioenvs/wemos-d1mini32/lib658/WiFi_ID870/WiFiClient.cpp.o
Compiling .pioenvs/wemos-d1mini32/lib658/WiFi_ID870/WiFiServer.cpp.o
Compiling .pioenvs/wemos-d1mini32/lib658/WiFi_ID870/WiFiUdp.cpp.o
Compiling .pioenvs/wemos-d1mini32/lib658/WiFi_ID870/utility/server_drv.cpp.o
```

xoseperez, 2018-10-15:
The APIs of the EEPROM_Rotate and EEPROM32_Rotate are different, that's true. I guess I could try to make them compatible up to some extent. But since the differences would be limited to the `eeprom.ino` file I guess it's OK to just add some build conditionals there on the setup method.

skarcha, 2018-10-15:
Ok, I'll try to work on it tonight. I'll start with why is not needed `.rotate()`, `.reserved()` and other methods in ESP32 version.

Zixim, 2018-12-05:
any movement on this?

sushantr5, 2019-10-01:
Any updates on ESP32 porting?

Tofandel, 2019-10-29:
Sadly, there was no PR or commit on this to take over.

After I finish the vue UI I might work on it because I need an esp32 port as well

Eta: 2~3 months before I have the time to dive in

lblabr, 2020-12-08:
are there any activities here ?

MikelMirror, 2021-10-04:
News?

MelanieT, 2022-01-14:
So where is the part that is already done?