xoseperez/espurna #690: Compilation with bme280 sensor not possible

kuppe234, 2018-03-17:
Hello

I would like to connect a bme280.
Unfortunately I get the following error when compiling:
```
/net/esl-strana-061/home/dhe/Projekte/Sonoff/espurna/code/espurna/settings.ino: In lambda function:
/net/esl-strana-061/home/dhe/Projekte/Sonoff/espurna/code/espurna/settings.ino:224:37: error: '_settingsI2CScanCommand' was not declared in this scope
_settingsI2CScanCommand();
^
*** [.pioenvs/itead-s20/src/espurna.ino.cpp.o] Error 1
```
How can I solve this?

Greetings, kuppe.

lobradov, 2018-03-17:
Which branch are you using? master or (latest) dev?

Try dev, I think there was some changes on this topic rather recently...

kuppe234, 2018-03-17:
Thank you, with branch dev it´s working.
Greetings, kuppe.

lobradov, 2018-03-17:
Great! Please close this issue if you no longer need help.

closed 2018-03-17 by kuppe234