xoseperez/espurna #692: Add an option to improve the RGB color management (MagicHome)

Emmegiemme, 2018-03-18:
Hello there,

I have loaded up the firmware 1.12.3 on my MagicHome LED Controller and all the functions are working fine, except an annoying snag on the red color. 
I have double checked the behavior on an identical controller, not yet flashed (so basically I was sending commands by the original APP) and the issue seems to be from a hardware point of view (not sure if it is on the controller or on the led strip).

Basically, if from the Colorpicker I pick up the colors "BLUE" and "GREEN", they will always show quite sharpe on my led strip, even if there is a small component of the other colors (eg. if you ask "#08FF08" you can still get a sharpe GREEN color and "#0808FF" still looking as BLUE).

Whereas only if the controller gets the color "#FF000" shows on the led strip a proper RED color, instead, when there is a very small component of the others colors (eg. "#FF0300") the led strip assumes a PINK color instead of the RED one.

As you can imagine, picking up a straight color from a Color picker ("#FF000") isn't an easy job, even if you point to the most RED color part, you will always get a small component of the others.

So, in order to avoid this annoying behavior, I would really love to have on the Espurna firmware an option to "ignore" the other components if they are lower than a set acceptable limit (eg. "#FF0505" forces to be "#FF0000").

Cheers
Mirco

xoseperez, 2018-03-19:
You can disable the color picker and use simple sliders to set each channel value.

xoseperez, 2018-07-02:
Closing this since there has been no feedback in several months. Please feel free to reopen if necessary.

closed 2018-07-02 by xoseperez