xoseperez/espurna #694: Try to build My own firmware for my Sonoff TH17 with SI7021 senaor but all failed

AlwinHummels, 2018-03-18:
I try and try but with no luck My PlatformIO gives 99-platformio-udev.rules errors on my Linux Mint System.
When I install the espurna-1.12.4-itead-sonoff-th.bin the sonof works fine but I can't use my sensor.

Can someone please help me to build a similar firmware with the SI7021 enabled?
Thanks any way. Hoe someone can help me build it my self but if someone already have a good working version the please ;-)

xoseperez, 2018-03-19:
Do you mean you cannot connect to the programmer? The platformio udev rules define the permissions for several programming (USB2UART) devices in the market.

AlwinHummels, 2018-03-19:
Did that and i'm a little further now. I'm a total nearby on this but like to learn.
What files do i have to edit so I have a equal version of the espurna-1.12.4-itead-sonoff-th.bin but with the SI7021 sensor enabled. With lines do I need to uncomment and what do i need to add to get the sensor working. Thanks for helping me so far

xoseperez, 2018-03-19:
The bare mininum is to enable support for this sensor. Look for `SI7021_SUPPORT` in `sensors.h`. While doing that check other available settings. By default it will try to find the address scanning the I2C bus (that's what the `SI7021_ADDRESS 0x00` does).

AlwinHummels, 2018-03-21:
Sorry for bugging you but I can't get it work. I setup platformio and downloaded the espurna mastercode. 
but witch files and what do I have to chanche to get a working firmware for my Sonoff TH16 with a Sonoff SI7021 sensor so I can use all with Domoticz. Just like the prebuild espurna-1.12.4-itead-sonoff-th.bin version. Thanks for all help.

mcspr, 2018-03-21:
@AlwinHummels If your goal is to have exactly the same file as releases, run this in the `code` directory of [current release](https://github.com/xoseperez/espurna/archive/1.12.4.zip)
```bash
$ pio run -e itead-sonoff-th
```
Resulting image file will be `code/.pioenvs/itead-sonoff-th/firmare.bin` (note the firmware is with **.bin**, not **.elf**)

As for *how* to add sensor, aforementioned file is `code/espurna/config/sensors.h` Just search for `SI7021_SUPPORT`

AlwinHummels, 2018-03-21:
I got `|-- <Hash> v1.0
Compiling .pioenvs/itead-sonoff-th/src/espurna.ino.cpp.o
Compiling .pioenvs/itead-sonoff-th/src/pwm.c.o
Archiving .pioenvs/itead-sonoff-th/lib884/libArduinoJson_ID64.a
Indexing .pioenvs/itead-sonoff-th/lib884/libArduinoJson_ID64.a
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFiAP.cpp.o

Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
/home/alwin/Downloads/espurna-1.12.4/code/espurna/settings.ino: In lambda function:
/home/alwin/Downloads/espurna-1.12.4/code/espurna/settings.ino:224:37: error: '_settingsI2CScanCommand' was not declared in this scope
_settingsI2CScanCommand();
^
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/WiFiClientSecure.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pioenvs/itead-sonoff-th/libfdf/ESP8266WiFi/WiFiUdp.cpp.o
*** [.pioenvs/itead-sonoff-th/src/espurna.ino.cpp.o] Error 1`

AlwinHummels, 2018-03-21:
I only changed sensors.h 
`//------------------------------------------------------------------------------
// SI7021 temperature & humidity sensor
// Enable support by passing SI7021_SUPPORT=1 build flag
//------------------------------------------------------------------------------

#ifndef SI7021_SUPPORT
#define SI7021_SUPPORT                  1
#endif

#ifndef SI7021_ADDRESS
#define SI7021_ADDRESS                  0x00    // 0x00 means auto
#endif

#if SI7021_SUPPORT
#undef I2C_SUPPORT
#define I2C_SUPPORT                     1
#endif

//----------------`

mcspr, 2018-03-21:
https://github.com/xoseperez/espurna/commit/17be19e55cbf0d9e7f627ee2f39f3f799f611275 from the upcoming version should do it.

AlwinHummels, 2018-03-21:
thanks build compleet now I gona test it now on my Sonof ;-)

AlwinHummels, 2018-03-23:
Flashed but still have DHT22 @ GPIO14 on my sensor in the webinterface and not My  SI7021_
I flashed with: 
` esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=8m -fm dout 0 firmware.bin`

mcspr, 2018-03-23:
Yeah, I forgot to mention the order of defines. `code/espurna/config/all.h` contains all of the included headers in specific order. And when `itead-sonoff-th` is used it implicitly enables DHT and DALLAS via `hardware.h` inside the `ITEAD_SONOFF_TH` section.

AlwinHummels, 2018-03-23:
On my hardware.h I edited:

`#elif defined(ITEAD_SONOFF_TH)

    // Info
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_TH"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1

    // Jack is connected to GPIO14 (and with a small hack to GPIO4)
    #ifndef DALLAS_SUPPORT
    #define DALLAS_SUPPORT      0
    #endif
    #define DALLAS_PIN          14

    #ifndef DHT_SUPPORT
    #define DHT_SUPPORT         0
    #endif
    #define DHT_PIN             14

    //#define I2C_SDA_PIN         4
    #define I2C_SCL_PIN         14`
Gives me now a firmware without any sensor ;-) but step by step I learn a lot ;-0

mcspr, 2018-03-23:
Ok. I was assuming that was i2c Si7021 board and not sonoff one. My understanding is that you already got it detected as DH22 (https://github.com/xoseperez/espurna/issues/216#issuecomment-360139341)

You said that you got DH displayed in the web. Doesn't it show the readings?

AlwinHummels, 2018-03-23:
The DH@gpio 14 shows nothing and it was with the espurna-1.12.4-itead-sonoff-th.bin but I can try it again with that version.

AlwinHummels, 2018-03-23:
from the debug I read: 
```
150189] [SENSOR]  -> ERROR 6
[150212] [SENSOR] Error reading data from DHT22 @ GPIO14 (error: 3)
[150317] [DOMOTICZ] Received value 0 for IDX 38
[150348] [DOMOTICZ] Received value 0 for IDX 37
[150378] [DOMOTICZ] Received value 0 for IDX 21
[150410] [DOMOTICZ] Received value 0 for IDX 55
[150442] [DOMOTICZ] Error parsing data
[150454] [DOMOTICZ] Error parsing data
[150815] [DOMOTICZ] Received value 0 for IDX 7
[151945] [DOMOTICZ] Received value 0 for IDX 723
[151972] [DOMOTICZ] Received value 0 for IDX 6
[152001] [DOMOTICZ] Received value 0 for IDX 724
[152030] [DOMOTICZ] Received value 0 for IDX 6
[152060] [DOMOTICZ] Received value 0 for IDX 725
[152077] [DOMOTICZ] Error parsing data
[152097] [DOMOTICZ] Error parsing data
[152133] [DOMOTICZ] Received value 0 for IDX 730
[153808] [DOMOTICZ] Received value 0 for IDX 729
[153836] [DOMOTICZ] Received value 0 for IDX 727
[153867] [DOMOTICZ] Received value 0 for IDX 728
[153897] [DOMOTICZ] Received value 0 for IDX 1
[153931] [DOMOTICZ] Received value 0 for IDX 2
[155403] [DOMOTICZ] Received value 0 for IDX 37
[156219] [SENSOR] Error reading data from DHT22 @ GPIO14 (error: 3)
[156804] [DOMOTICZ] Received value 1 for IDX 735
[156834] [DOMOTICZ] Received value 0 for IDX 38
[156867] [DOMOTICZ] Received value 0 for IDX 37
```

AlwinHummels, 2018-03-23:
https://www.itead.cc/wiki/Sonoff_Sensor_Si7021 that's the sensor

xoseperez, 2018-03-24:
@AlwinHummels I have this sensor and it is working fine for me. The error 3 is a timeout in the communication, so either the GPIO is not right (but it should be 14 for Sonoff TH) or the sensor is faulty. Mind that this specific sensors (not SI7021 in general, but Itead's one) is not and I2C sensor. Just disable the I2C lines and the DALLAS lines in the hardware.h file to remove possible incompatibilities and test again. 

Also, copy here the log on boot, please.

xoseperez, 2018-07-02:
@AlwinHummels did you fixed the issue?

closed 2018-09-07 by stale[bot]