xoseperez/espurna #695: Add support of dual white/cct led strips (H801)

Meinsss, 2018-03-18:
Hi there,

thanks for your awesome work!
I wanted to use your firmware to control my cct led strips (24V+, CW, WW) with the H801.
When working with Home Assistant (MQTT) I can only work with the _color_temp_command_topic_ when under _Lights>Use color_ is enabled. Unfortunately the values are getting sent on Channel 0/1/2 (RGB) and it would be nice to have a feature to get the values for Cold White and Warm White (on 2 channels).

Example:
[image]
Cold White/Channel 1: 255
Warm White/Channel 2: 0

Skaronator, 2018-04-01:
The current RGBW implementation isn't that good either. I've enabled this setting in the webinterface:
[image]

And when I set it in the espurna webinterface to perfect white it'll disable the RGB strip but if it isn't perfectly white it will still show the RGB strip at 100% brightness.

@xoseperez You need to convert the RGB value in the webinterface to a RGBW value. Here is a example code in python:
https://github.com/home-assistant/home-assistant/blob/98e4d514a5130b747112cc0788fc2ef1d8e687c9/homeassistant/util/color.py#L372-L381

It is basically get the lowest number and subtract it from all numbers and use it as white value.

Skaronator, 2018-04-01:
Okay just testing with H801 via MQTT in Home Assistant and IMO the `white_value_state_topic` and `white_value_command_topic` command must be removed to get rid of this slider:

[image]

And the white value must be extracted from the RGB value and when you have the white color value then you can easily calculate the color temperature to set W1 between 0-255 and W2 between 0-255.

I can help you with the maths if you want :P

soprater, 2018-04-03:
The icing on the cake would be to "abuse" controllers with more than 4 channels (e.g. H801) to use 2 cct strips at once (2x warm white, 2x cold white). I'd definitely help, but unfortunately I can't program c.

Skaronator, 2018-04-03:
@soprater 
You can do this already... just disable the RGB thing and control each channel on its own.
[image]

soprater, 2018-04-03:
I mean like receiving color temperatures on 2 different topics (think it's in mired from HA)

soprater, 2018-04-04:
#754 
Tried my best to support this project.

PR #785 (2018-05-13): Implement CCT warm white and cold white LEDs for devices with 5 channels.
Tested and it works fine. **Only thing that is missing is a slider to set the Mired value in the Webinterface.**

The current approach is to use the white value of the RGB to RGBW calculation and split it in warmwhite and coldwhite based on the Mireds (153-500). It works fine but is not perfect. We just assume that you have a 2000 Kelvin and 6500 Kelvin LEDs. I might improve this in a future PR so we can include the RGB strip but we'll see.

Close #754
Fix #695
closed 2018-05-13 by xoseperez