xoseperez/espurna #696: SonOff POW Bad Power Measurement

HugoMSMartins, 2018-03-19:
Helo,
I have the SonOff POW with 1.12.4 Espurna FW, connected to Solar Panels, to measure the instant generated power and the cumulative energy.

During the night without sun, i have many peaks the power goes to 200W, 800W, 1000W etc where it should give 0 or 1 W.

Anything is not good in the consumption measurement. seems to worsen when the consumption is close to 0W, low values.

HugoMSMartins, 2018-03-19:
[image]

AlbertWeterings, 2018-03-30:
SonOff POW is not using the HLW8012 chip anymore but the cse7759

Question is what is the difference

gn0st1c, 2018-03-30:
cse7759 and hlw8012 are the same (pins, meter calculations etc)

i didn't know that sonoff switched to cse. thank you for the information :)

AlbertWeterings, 2018-03-31:
Comparing data sheets of [CSE7759](https://datasheet.lcsc.com/szlcsc/CSE7759_C126724.pdf) and [HLW8012](https://datasheet.lcsc.com/szlcsc/HLW8012_C83804.pdf)

There is a significant difference in the data sheets between CSE7759 and HLW8012 The most important difference is a **note** that is in the CSE7759 datasheet and not in the HLW8012 this may be the reason for measuring energy when no load is connected [See Page 10.](https://datasheet.lcsc.com/szlcsc/CSE7759_C126724.pdf)

> Note: Due to the noise of CSE7759, **if there is no signal input, there will be about 2Hz signal output on the CF1 pin** indicating the current and voltage rms value (this signal can be judged by the CF pulse frequency to hide the movement). To ensure the correct measurement of the effective value, the effective value of the current applied to the channel V1P\V1N shall not be less than 40uv, and the effective value of the voltage applied to the channel V2P shall not be less than 480uv. When using, please consult our technical support.

A other difference between both is also found in the **Limit rating** page 8 (CSE7759) page 6 (HLW8012) of both datasheets. It is the Max voltage between V1P, V1N, V2P for HLW8012 it is -2/+2 volt for CSE7759 it is -1/+1 volt.

Last difference is found on page 9 (CSE7759) page 7 (HLW8012). It is the application diagram. CSE7759 has on pin 4 (VP2) 2 resistors and 1 capacitors. The two resistors are a voltage divider set to 1000/1 Where the HLW8012 has the same circuit build up out of 6 x 470K and 1K resistors making a voltage divider set to 2820/1.

I think it would be good to know from people who are reporting strange energy measurements the exact type of chip installed in the SonOff Pow.

gn0st1c, 2018-03-31:
i had the [Rev.1.0 datasheet of cse7759](http://www.datasheetspdf.com/pdf/1047926/CHIPSEA/CSE7759/1)

* limit rating would not effect us.
* resistors we can set with HLW8012_CURRENT_R, HLW8012_VOLTAGE_R_UP and HLW8012_VOLTAGE_R_DOWN
* but the noise is a big deal.

AlbertWeterings, 2018-03-31:
@gn0st1c knowing a bit of Chinese helps to find difference in datasheets, after that google is my best friend in getting the details 😁. Resistors is not what we have to worry about both chips do calculations in the exact same way by the same formula and our friends at itead did not change anything to the resistors when they changed from HLW8012 to CSE7759.

What we need to know for sure is if this issue only occurs with sonoff pow units having the CSE7759. Or better if this issue is with both chips but more extreme as with HLW8021.

If the 2hz signal only occurs when no load is applied and the signal has a complete different frequency when load is applied it must be easy to filter the 2hz signal out. Hopefully filtering the signal away is not affecting the working of the HLW8012 otherwise we get two different versions of firmware for the same device and users first have to check what chip is in it.

gn0st1c, 2018-04-01:
i think what note means is, if i understand correctly;

we use CF for power consumption. if there is no power consumption, ignore CF1 (for voltage and current)

AlbertWeterings, 2018-04-02:
@gn0st1c I think you read the note correct We can try this I have a few in-used Sonoff Pow units to test.
But there is something more in the note and I think this is why for CSE5579 the resistors on pin V2P are different as with HLW8012 the voltage om pin V2P must not fall below 480uV. With the resistors as with HLW8012 the voltage is divided 2820/1 and for CSE5579 1000/1 so the using the resistors as in the CSE5579 datasheet will result in a higher voltage on pin V2P. So we also have to find out if the V2P voltage drops below 480uV when the higher resistors are used.

gn0st1c, 2018-04-03:
they also have CSE7759 B..
damn, too many sensors, too little time :)

[CSE 7756 with uart][http://www.chipsea.com/UploadFiles/2017/08/11144342F01B5662.pdf]
[CSE 7758](http://www.chipsea.com/UploadFiles/2013/01/07151240691B2665.pdf)
[CSE 7759 B with uart](http://www.chipsea.com/UploadFiles/2018/02/0809255992C2F56E.pdf)

AlbertWeterings, 2018-04-03:
I think go by little steps first solve issues with the new chip used in sonoff pow😁