xoseperez/espurna #700: Internal Schedule to set particular Colour / Dimmer setting rather than just ON/OFF

oxcxsxixbxaxn, 2018-03-20:
The built in scheduler is a great addition and would like to suggest request some improvements.
While I have several sOnOff basic switches I mostly use RGB/RGBW lighting and it would be very useful if the internal scheduler could be set to different dimmer / colour levels and maybe even a local device API command.

wysiwyng, 2018-03-20:
PWM scheduler support was added to the dev branch with commit 8777d29f7c780bc14396a795c947d777b6d97262. You might need to compile the firmware yourself; see the wiki for excellent instructions; or wait until a new release including this feature comes out.

oxcxsxixbxaxn, 2018-03-22:
Oh good, seems like what I was looking for.  I compile myself but so far been sticking to stable branch.  I'll take a look.  

Thanks

oxcxsxixbxaxn, 2018-05-06:
Now implemented as dimmable and switchable per channel 0-3.
released as 1.2.6

closed 2018-05-06 by oxcxsxixbxaxn