xoseperez/espurna #702: SONOFF RF bridge reboots itself

akasma74, 2018-03-21:
Flashed my RF bridge yesterday with the current Espurna, and in less than 24 hours it suddenly rebooted itself:

home/sonoff_rfbridge/rfin 30A201A404C4D459CE
home/sonoff_rfbridge/relay/0 1
home/sonoff_rfbridge/rfin 30CA01A404C4D4AB2E
home/sonoff_rfbridge/relay/1 1
home/sonoff_rfbridge/relay/0/set 0
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/relay/1/set 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/app ESPURNA
home/sonoff_rfbridge/version 1.12.5a
home/sonoff_rfbridge/host srfb-5121
home/sonoff_rfbridge/ip 192.168.0.35
home/sonoff_rfbridge/mac DC:4F:22:82:51:21
home/sonoff_rfbridge/rssi -38
home/sonoff_rfbridge/uptime 59474
home/sonoff_rfbridge/datetime 2018-03-21 08:51:01
home/sonoff_rfbridge/freeheap 18328
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/relay/2 0
home/sonoff_rfbridge/relay/3 0
home/sonoff_rfbridge/relay/4 0
home/sonoff_rfbridge/relay/5 0
home/sonoff_rfbridge/relay/6 0
home/sonoff_rfbridge/relay/7 0
home/sonoff_rfbridge/vcc 3410
home/sonoff_rfbridge/status 1
home/sonoff_rfbridge/loadavg 2
home/sonoff_rfbridge/app ESPURNA
home/sonoff_rfbridge/version 1.12.5a
home/sonoff_rfbridge/host srfb-5121
home/sonoff_rfbridge/ip 192.168.0.35
home/sonoff_rfbridge/mac DC:4F:22:82:51:21
home/sonoff_rfbridge/rssi -37
home/sonoff_rfbridge/uptime 59774
home/sonoff_rfbridge/datetime 2018-03-21 08:56:01
home/sonoff_rfbridge/freeheap 18328
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/relay/2 0
home/sonoff_rfbridge/relay/3 0
home/sonoff_rfbridge/relay/4 0
home/sonoff_rfbridge/relay/5 0
home/sonoff_rfbridge/relay/6 0
home/sonoff_rfbridge/relay/7 0
home/sonoff_rfbridge/vcc 3399
home/sonoff_rfbridge/status 1
home/sonoff_rfbridge/loadavg 2
home/sonoff_rfbridge/app ESPURNA
home/sonoff_rfbridge/version 1.12.5a
home/sonoff_rfbridge/host srfb-5121
home/sonoff_rfbridge/ip 192.168.0.35
home/sonoff_rfbridge/mac DC:4F:22:82:51:21
home/sonoff_rfbridge/rssi -38
home/sonoff_rfbridge/uptime 60074
home/sonoff_rfbridge/datetime 2018-03-21 09:01:01
home/sonoff_rfbridge/freeheap 18328
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/relay/2 0
home/sonoff_rfbridge/relay/3 0
home/sonoff_rfbridge/relay/4 0
home/sonoff_rfbridge/relay/5 0
home/sonoff_rfbridge/relay/6 0
home/sonoff_rfbridge/relay/7 0
home/sonoff_rfbridge/vcc 3410
home/sonoff_rfbridge/status 1
home/sonoff_rfbridge/loadavg 2
home/sonoff_rfbridge/rfin 30B601A404C4D459CE
home/sonoff_rfbridge/relay/0 1
home/sonoff_rfbridge/relay/0/set 0
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/rfin 30D401A404C4D4AB2E
home/sonoff_rfbridge/relay/1 1
home/sonoff_rfbridge/relay/1/set 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/status 0
home/sonoff_rfbridge/app ESPURNA
home/sonoff_rfbridge/version 1.12.5a
home/sonoff_rfbridge/host srfb-5121
home/sonoff_rfbridge/ip 192.168.0.35
home/sonoff_rfbridge/mac DC:4F:22:82:51:21
home/sonoff_rfbridge/rssi -41
home/sonoff_rfbridge/uptime 7
home/sonoff_rfbridge/freeheap 17504
home/sonoff_rfbridge/relay/0 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/relay/2 0
home/sonoff_rfbridge/relay/3 0
home/sonoff_rfbridge/relay/4 0
home/sonoff_rfbridge/relay/5 0
home/sonoff_rfbridge/relay/6 0
home/sonoff_rfbridge/relay/7 0
home/sonoff_rfbridge/vcc 3394
home/sonoff_rfbridge/status 1
home/sonoff_rfbridge/loadavg 100
home/sonoff_rfbridge/rfin 30CA01A404C4D4AB2E
home/sonoff_rfbridge/relay/1 1
home/sonoff_rfbridge/relay/1/set 0
home/sonoff_rfbridge/relay/1 0
home/sonoff_rfbridge/rfin 311A01A404C4D4B2DE
home/sonoff_rfbridge/relay/4 1
home/sonoff_rfbridge/relay/4/set 0
home/sonoff_rfbridge/relay/4 0
home/sonoff_rfbridge/rfin 30C001A404C4D51C2E
home/sonoff_rfbridge/relay/5 1
home/sonoff_rfbridge/relay/5/set 0
home/sonoff_rfbridge/relay/5 0
home/sonoff_rfbridge/rfin 30C001A404C4D459CE
home/sonoff_rfbridge/relay/0 1
home/sonoff_rfbridge/relay/0/set 0
home/sonoff_rfbridge/relay/0 0

What was it? That loadavg 100 is strange, isn't it?

xoseperez, 2018-03-24:
Load average is calculated based on the device own history doing a simple count. So 100% means at that moment it was busy as never before since last reboot. Not necessarily a bad thing.

To debug the issue it would be interesting to have the debug log. If you are still using ESPurna you can log via telnet to the device IP and monitor the log.

akasma74, 2018-03-24:
I will try and report back

mcspr, 2018-03-25:
@akasma74 You need to enable it first via `TELNET_STA` define or web **Admin > Enable TELNET**

rfbridge build disables terminal commands so telnet is read-only debug log

akasma74, 2018-03-26:
@mcspr Max, I enabled telnet via web interface and rebooted the device, but still get `Connection closed by foreign host.`
Any ideas how to overcome it?

xoseperez, 2018-03-26:
Did you save the settings before rebooting?

akasma74, 2018-03-26:
@xoseperez Of course I did, and double-checked after rebooting - it shows telnet enabled on web interface, but still no joy with telnet, it apparently times out

mcspr, 2018-03-26:
@akasma74 I would also check if it works at all in AP mode.
Another thing possible that there is already open connection as telnet module limits connections:
https://github.com/xoseperez/espurna/blob/8f0af3eb20c873dd620cff2a50c94de7c71ef47e/code/espurna/config/general.h#L31

akasma74, 2018-03-26:
@mcspr What's AP mode and how to check it?
I assume that if I reboot the bridge, ALL connections are dropped, and I rebooted it and then tried to telnet wit no result.

mcspr, 2018-03-26:
Right now bridge has configured WiFi **A**ccess **P**oint ssid and connects to it. But device can make itself WiFi access point that you can connect to with a phone or laptop and configure it.

You can force it by deleting all wireless networks from WiFi tab and rebooting. It will show up as **ESPURNA_something** with admin password as passphrase. 192.168.4.1 will be the ip address

xoseperez, 2018-07-02:
@akasma74 did you manage to get more info about the reboots?

akasma74, 2018-07-02:
Hi Xose,
I didn't as I couldn't establish a telnet connection with my RFBridge.
It works fine in AP mode though.
Could you suggest any other ways to try?
I haven't updated the firmware for a while, but can give it a go as it's not in daily use anymore (using Arduino Mega + RFLink instead).

xoseperez, 2018-07-02:
Yes, it's always a good idea to report on the latest released version (1.13.0 for the time being). Thank you.

closed 2018-09-07 by stale[bot]