xoseperez/espurna #703: Procedure to add custom IR Remotes

darshkpatel, 2018-03-21:
I wanted to add a custom IR remote to switch Relays, currently running custom code on my esp to switch a 8 channel relay , wanted to migrate to espurna but couldn't figure out how to add my IR remote 

I couldn't find relevant Documentation in the wiki, so which files do i need to modify to add IR codes for my remote and assign them to switches ? could you please point me in the right direction ?

Also , I'd love to update the documentation for adding custom remotes after i figure it out.

xoseperez, 2018-03-24:
There is no documentation about that yet, sorry. You can investigate the RF section in `code/espurna/config/general.h` where there are two remotes defined. I think it's pretty straight forward. Notice there is some wishful thinking there, like the IR_BUTTON_MODE_EFFECT mode which is not yet implemented.

If you manage to add your remote please issue a PR and it will be merged to the code. Also, you can update the wiki yourself by using the edit buttons.

gn0st1c, 2018-03-25:
@darshkpatel : use [irdumpv2](https://github.com/markszabo/IRremoteESP8266/tree/master/examples/IRrecvDumpV2) to decode your ir remote.

it will generate something like this;
>Timestamp : 002222.937
>Encoding  : NEC
>Code      : FFE01F (32 bits)
>Library   : v2.3.2
>
>Raw Timing[71]:
>   + 10816, -  4494,    +   674, -   572,    +   674, -   572,    +   672, -   572,
>   +   672, -   572,    +   672, -   600,    +   646, -   572,    +   672, -   572,
>   +   672, -   572,    +   676, -  1678,    +   668, -  1680,    +   666, -  1708,
>   +   644, -  1682,    +   670, -  1682,    +   672, -  1680,    +   678, -  1682,
>   +   674, -  1682,    +   676, -  1704,    +   658, -  1680,    +   682, -  1682,
>   +   686, -   574,    +   684, -   574,    +   682, -   574,    +   744, -   518,
>   +   670, -   570,    +   668, -   570,    +   672, -   572,    +   676, -   572,
>   +   676, -  1678,    +   678, -  1680,    +   676, -  1680,    +   676, -  1680,
>   +   680, -  1682,    +   746, - 40008,    + 10806, -  2228,    +   668
>
>uint16_t rawData[71] = {10816, 4494,  674, 572,  674, 572,  672, 572,  672, 572,  672, 600,  646, 572,  672, 572,  672, 572,  676, 1678,  668, 1680,  666, 1708,  644, 1682,  670, 1682,  672, 1680,  678, 1682,  674, 1682,  676, 1704,  658, 1680,  682, 1682,  686, 574,  684, 574,  682, 574,  744, 518,  670, 570,  668, 570,  672, 572,  676, 572,  676, 1678,  678, 1680,  676, 1680,  676, 1680,  680, 1682,  746, >40008,  10806, 2228,  668};  // NEC FFE01F
>uint32_t address = 0x0;
>uint32_t command = 0x7;
>uint64_t data = 0xFFE01F;

you can use the uint64_t data = 0xFFE01F value.

general.h;
>        { 0xFFE01F, IR_BUTTON_MODE_STATE, 1 }, // on

darshkpatel, 2018-03-25:
@gn0st1c @xoseperez thanks ,
I'd work on it around April 23rd. 

Would issue a PR and add documentation for the same ✌️

darshkpatel, 2018-05-02:
solved.
updated the wiki too

closed 2018-05-02 by darshkpatel