xoseperez/espurna #704: wiki missing images

gn0st1c, 2018-03-21:
missing device images, they were available before (when we merged my wiki with yours)
ManCaveMade ESP-Live, InterMitTech QuinLED 2.6, Xenon SM-PW702U

also can you upload these images (for V9261F and ECH1560 flashing)
[generic_powers_flashing.zip](https://github.com/xoseperez/espurna/files/1833566/generic_powers_flashing.zip)

gn0st1c, 2018-03-21:
Oh shoot! images of those devices are in espurna repository but not in wiki rep.
can you copy the devices folder to wiki ?

gn0st1c, 2018-03-22:
can you also upload these as well?
* tonbux-xs-ssa06 device and flashing files.
* extra board information for previous devices.

[wiki-leaks.zip](https://github.com/xoseperez/espurna/files/1837605/wiki-leaks.zip)

gn0st1c, 2018-03-22:
let me know when you upload generic_powers_flashing.zip & wiki-leaks.zip so that i can edit the wiki.

don't forget to sync espurna devices folder with wiki devices folder.

xoseperez, 2018-03-24:
Uploaded & synced.
Sorry, I coudn't do it before. I'm busy lately.

closed 2018-03-24 by xoseperez

gn0st1c, 2018-03-24:
no worries, updated the wiki.