xoseperez/espurna #705: NTP compile issues

zafrirron, 2018-03-21:
ntp:40: error: 'class NTPClient' has no member named 'setNTPTimeout'
ntp:71: error: 'class NTPClient' has no member named 'setDSTZone'

NTP library:
name=NtpClientLib
version=2.5.0
author=German Martin
maintainer=German Martin
sentence=Ntp Client Library
paragraph=Library to get system sync from a NTP server. Based on code from NTP client example. Currently, it works on ESP8266 based boards. I've made it compatible with Arduino boards w Eternet module but I have not had the opportunity to test it. Please, add an issue to GitHub if you find a bug. NOTICE: After version 2.0.0 library structure has changed. Please refer to README file on github repository.
category=Timing
url=https://github.com/gmag11/NtpClient
architectures=*

AlbertWeterings, 2018-03-21:
This is due to New version of the NTP library by German it is not yet compatible with ESPURNA version 1.12.4

zafrirron, 2018-03-22:
thanks @AlbertWeterings,
Any idea what NTP library version should I use?

AlbertWeterings, 2018-03-22:
@zafrirron This one is still working on my side. https://github.com/gmag11/NtpClient.git#28af3a5

zafrirron, 2018-03-22:
Thanks @AlbertWeterings

closed 2018-03-22 by zafrirron

danriches, 2018-03-22:
I tried to use the library you specified above and I get compile errors for the following missing functions:
setDSTZone
setNTPTimeout

I've commented them out in the code for the moment but this needs looking at, Thanks!

AlbertWeterings, 2018-03-23:
Then you are using still the wrong library I think. Did you uninstall the previous installed Lib?
In Arduino Library Manager search for ntpclient. This is the one installed on my side.
[image]

danriches, 2018-03-23:
Well I deleted the library folder in c:\users\daniel\documents\arduino\libraries and checked the compilation failed due to a missing library. Then I installed v2.5.0 as you've said above and I checked the file was by German Martin, his name is in the cpp and header file. Still it fails to compile for those two functions I've stated. I checked the compile output and indeed it is referencing the new library; "-IC:\Users\Daniel\Documents\Arduino\libraries\NtpClientLib\src"

So what to do, the German Martin v2.5.0 library does not have those functions so it won't compile. I don't quite understand then how you've got it to compile. I've scanned my whole computer and there is no other NtpClientLib it can use. Please would you mind checking your compilation string and verifying that that is the library to use please?

Thanks and sorry to be a pain, Dan

AlbertWeterings, 2018-03-23:
@danriches Are yoy trying to compile the dev. or master branch of ESPURNA? The master is compiling on my side using this and all libraries mentioned in [platformio.ini](https://github.com/xoseperez/espurna/blob/dev/code/platformio.ini) and the [Using the Arduino IDE](https://github.com/xoseperez/espurna/wiki/ArduinoIDE)

danriches, 2018-03-23:
Ah yes, that'll be why duh stupid me! I've downloaded the V1.12.4 source and copied my settings over bit by bit using WinMerge carefully and it's compiled first time. My hat goes off to Xose and all the developers that contributed for great code, ease of use and sheer number of device support, awesome!!

Thanks ever so much @AlbertWeterings

AlbertWeterings, 2018-03-23:
No problem Dan if we can help each other why not 😁