xoseperez/espurna #709: Wemos D1 R1 hangs after 15 minute of operation

afefelov, 2018-03-22:
Hello Jose, first of all thank you for your perfect work.

I'm installed epsurna 1.12.4 on my wemos d1 r1 (old version from 2015).
I connected ds1820 to gpio4, led to gpio 13 and button to gpio 5 and ground.

```

    // Buttons
    // No buttons on the D1 MINI alone, but defining it without adding a button doen't create problems
    #define BUTTON1_PIN         0  // Connect a pushbutton between D3 and GND,                                // it's the same as using a Wemos one button shield
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON2_PIN         5   
    #define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

// LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    0
    #define LED2_PIN            13
    #define LED2_MODE           LED_MODE_MQTT
    #define LED2_PIN_INVERSE    0
```

Well it is starting, connecting to wifi, sending data to mosquitto, and even get data back (toggling led2), but, after 10-15 minutes it hangs completely, internal led which is normally blinking (LED_MODE_WIFI) just staying ON, no data in serial and network also down.

I was wondered if it is due to weak power unit and weak wifi, I put it on USB port, and leave in 5 meters from my mikrotik, and it is still hangs.

I wish to find out what exactly is going on and pay my 50cent to community for efforts, I've read a bit about debugging esp8266 but it seems there is nothing related to situation when serial is not available. 

Could you please, let me know how can I get debug info for further analysis? I'm also wish to flash another one d1 r1 today, just to exclude situation when there is a problem with given sample of hw.

Thanks.

xoseperez, 2018-03-24:
Have you managed to get the debug log when the device hangs? It might be useful to know what was it doing...

afefelov, 2018-03-27:
Hi @xoseperez, I'm a newbie and I don't know how to get debug log from hanged device.
I've flashed 2 devices with exact the same firmware and got exactly same behaviour. Later on I disabled mds(server/client), homeassistant and scheduler. And it working now well stable.

What I can suppose: it seems that mdns conflicts somehow with my mikrotik dns server somehow I will go on investigation on what's going on.

xoseperez, 2018-04-03:
Hi @afefelov, that's interesting. I have no evidence the mDNS might be clashing with other DNS, I don't even think that's possible since mDNS is decentralized.

Current dev branch is not stable and I'm debugging the issue. At the moment my only guess is that the new NTPClientLib revision is blocking the watchdog. I'm also reverting some configuration changes in the WiFi connection (this looks more promising).

afefelov, 2018-04-03:
Well, I also had avahi-daemon at my local Ubuntu, disabled it and will build espurna again today.

Smartype, 2018-04-03:
I have several  itead-s22 and nodemcu boards with the same issue. No network or serial output. watchdog timer does not fire. It's very hard to debug without serial connection. Do you have any suggestions on how to find where it hangs?

Try to enable all the debug_flags in platformio.ini, and did not find any cue and it hangs in somewhere no log outputs.

This happens to arduino 2.4.0 and 2.4.1 only, So I do not know it's the esp8266 arduino framework issue.. 

I have not see such issue with 2.4.1-rc2 yet (But with other issues like crashing).

Smartype, 2018-04-03:
FYI, I have web removed and enabled mqtt with ssl. Telnet is used in AP mode to upload config.(KEYS/SET commands)

xoseperez, 2018-04-04:
Another suspect is lwIP. These changed in 2.4.0 to use version 2.0 and it is known to be stable.I think 2.4.1 solves the problem thou. Anyway try building 2.4.0 with lwIP 1.4. Mind that the prebuild binaries I release still use Core 2.3.0 and this is one of the reasons.

afefelov, 2018-04-06:
BTW, I've also built successful flash with 2.3.0 as well as disabling mdns, guys.

afefelov, 2018-05-11:
Well, I 've successfully built it with 2.4.1+IwIP 1.4

xoseperez, 2018-05-13:
Are you still experiencing problems with 2.4.X and lwIP 1.4?

nunoronconcouto, 2018-07-30:
What is the best firmware for Wemos D1 R2? 
Thanks

afefelov, 2018-08-03:
After 3 month of usage with power outages I can confirm that current setup is quite reliable: 2.3.0, mdns is off,  IwIP is 1.4.

Same setup is working stable on Wemos mini.

Issue should be closed, thanks for all your help.

closed 2018-08-03 by afefelov