xoseperez/espurna #715: MQTT disconnects randomly

richlock1, 2018-03-23:
First of all thanks for this very great project. The best framework for ESP8266.
But I have an instability issue. The main problem that sometimes the MQTT conection is disconected and the reconnection takes 1 minutes to 15 minutes.
The frequency is 20 times/day. During reconnect phase I miss data. The mosquitto broker is always available as from the logs other clients have sucessfull publishes.
The problem comes on Wemos D1 mini board with DHT22 sensor and with DS18B20 temperature sensor and also on Electrodragon 2 ch relay board without any sensor.
I saw the same behaviour with version 1.12.3 and now with 1.12.4.
On both card telnet is enabled, on Wemos thingspeak feature is on and I saw missing data on their web interface.
Earlier I used an free RTOS based fw on the wemos, and there was not missing thingspeak data, so I guess the hardwares are OK.

I logged the electrodragon board and here I show, that the MQTT disconnects random period. During the same time I also pingged that device, and that is always working. 
Sometimes the ping fails with 1 per minute frequency, but I think the issue is somewhere else as MQTT reconnect takes in bad case 15 minutes.

Here is an example when it takes 4 minutes to reconnect again.

[067340] [INIT] ESPURNA 1.12.4
[067340] [INIT] xose.perez@gmail.com
[067340] [INIT] http://tinkerman.cat

[067340] [INIT] CPU chip ID: 0x21C176
[067342] [INIT] CPU frequency: 80 MHz
[067345] [INIT] SDK version: 1.5.3(aec24ac9)
[067349] [INIT] Core version: 2_3_0
[067352] [INIT] Core revision: 
[067355] 
[067
[067360] [INIT] Flash speed: 40000000 Hz
[067363] [INIT] Flash mode: DIO
[067366] 
[067367] [INIT] Flash sector size:     4096 bytes
[067371] [INIT] Flash size (CHIP):  4194304 bytes
[067375] [INIT] Flash size (SDK):   4194304 bytes / 1024 sectors
[067381] [INIT] Firmware size:       465120 bytes /  114 sectors
[067387] [INIT] OTA size:           26  654 sectors
[067392] [INIT] SPIFFS size:              0 bytes /    0 sectors
[067398] [INIT] EEPROM size:           4096 bytes /    1 sectors
[067404] [INIT] Empty space:          16384 bytes /    4 sectors
[067409] 
[067410] [INIT] BUILD_FLAGS: 
[067413] [INIT] BOARD: ELECTRODRAGON_WIFI_IOT
[067417] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[067429] External System
[067434] [INIT] Settings size: 1268 bytes
[067438] [INIT] Free heap: 25064 bytes
[067441] [INIT] Power: 3156 mV
[067443] [INIT] Power saving delay value: 10 ms
[067448] 
[067448] [WIFI] MODE STA -------------------------------------
[067454] [WIFI] SSID  DIGI-UW83
[067457] [WIFI] IP    192.168.0.31
[067460] [WIFI] MAC   DC:4F:22:21:C1:76
[067463] [WIFI] GW    192.18.0.1
[067466] [WIFI] DNS   192.168.0.1
[067469] [WIFI] MASK  255.255.255.0
[067472] [WIFI] HOST  ESPURNA_21C176
[067475] [WIFI] BSSID 34:B3:54:BB:45:28
[067479] [WIFI] CH    2
[067481] [WIFI] RSSI  -64
[067483] [WIFI] ----------------------------------------------
[067489] +OK
[307855] [MQTT] Sending ESPURNA_21C176/app => ESPURNA (PID 1)
[307856] [MQTT] Sending ESPURNA_21C176/version => 1.12.4 (PID 1)
[307856] [MQTT] Sending ESPURNA_21C176/host => ESPURNA_21C176 (PID 1)
[307862] [MQTT] Sending ESPURNA_21C176/ip => 192.168.0.31 (PID 1)
[307867] [MQTT] Sending ESPURNA_21C176/mac => DC:4F:22:21:C1:76 (PID 1)
[307874] [MQTT] Sending ESPURNA_21C176/rssi => -65 (PID 1)
[307879] [MQTT] Sending ESPURNA_21C176/uptime => 307 (PID 1)
[307884] [MQTT] Sending ESPURNA_21C176/datetime => 2018-03-19 21:02:03 (PID 1)
[307891] [MQTT] Sending ESPURNA_21C176/freeheap => 25040 (PID 1)
[307897] [MQTT] Sending ESPURNA_21C176/relay/0 => 0 (PID 1)
[307902] [MQTT] Sending ESPURNA_21C176/relay/NA_21C176/vcc => 3148 (PID 1)
[... 311 lines cut ...]
--- 192.168.0.31 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 106.702/106.702/106.702/0.000 ms
2018. márc. 20., kedd, 02:39:07 CET
PING 192.168.0.31 (192.168.0.31) 56(84) bytes of data.
64 bytes from 192.168.0.31: icmp_seq=1 ttl=127 time=1.80 ms

--- 192.168.0.31 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 1.806/1.806/1.806/0.000 ms
2018. márc. 20., kedd, 02:39:08 CET
PING 192.168.0.31 (192.168.0.31) 56(84) bytes of data.
64 bytes from 192.168.0.31: icmp_seq=1 ttl=127 time=1.75 ms

--- 192.168.0.31 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 1.757/1.757/1.757/0.000 ms
2018. márc. 20., kedd, 02:39:09 CET
PING 192.168.0.31 (192.168.0.31) 56(84) bytes of data.
64 bytes from 192.168.0.31: icmp_seq=1 ttl=127 time=849 ms

..............

Coud you give me some hints what to set, what to debug more?
Thanks.

xoseperez, 2018-03-24:
What do the logs say on the mosquitto side?

richlock1, 2018-03-24:
It says:

Mar 20 02:36:41 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:36:41 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu
...
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:37:02 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/app', ... (7 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/version', ... (6 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/host', ... (14 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/ip', ... (12 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/mac', ... (17 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/rssi', ... (3 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/uptime', ... (5 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/datetime', ... (19 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/freeheap', ... (5 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/relay/0', ... (1 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/relay/1', ... (1 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/vcc', ... (4 bytes))
Mar 20 02:37:02 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/status', ... (1 bytes))
...
Mar 20 02:37:26 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:37:26 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu
...

Mar 20 02:38:11 raspberrypi mosquitto[531]: Client GarzsKapu has exceeded timeout, disconnecting.
.....
Mar 20 02:42:06 raspberrypi mosquitto[531]: New client connected from 192.168.0.31 as GarzsKapu (c0, k30).
Mar 20 02:42:06 raspberrypi mosquitto[531]: Sending CONNACK to GarzsKapu (0)
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received UNSUBSCRIBE from GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: #011#
Mar 20 02:42:06 raspberrypi mosquitto[531]: GarzsKapu #
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received SUBSCRIBE from GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: #011ESPURNA_21C176/relay/+/set (QoS 0)
Mar 20 02:42:06 raspberrypi mosquitto[531]: GarzsKapu 0 ESPURNA_21C176/relay/+/set
Mar 20 02:42:06 raspberrypi mosquitto[531]: Sending SUBACK to GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received SUBSCRIBE from GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: #011ESPURNA_21C176/led/+/set (QoS 0)
Mar 20 02:42:06 raspberrypi mosquitto[531]: GarzsKapu 0 ESPURNA_21C176/led/+/set
Mar 20 02:42:06 raspberrypi mosquitto[531]: Sending SUBACK to GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received SUBSCRIBE from GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: #011ESPURNA_21C176/action/set (QoS 0)
Mar 20 02:42:06 raspberrypi mosquitto[531]: GarzsKapu 0 ESPURNA_21C176/action/set
Mar 20 02:42:06 raspberrypi mosquitto[531]: Sending SUBACK to GarzsKapu
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/app', ... (7 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/version', ... (6 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/host', ... (14 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/ip', ... (12 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/mac', ... (17 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/rssi', ... (3 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/uptime', ... (5 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/datetime', ... (19 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/freeheap', ... (5 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/relay/0', ... (1 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/relay/1', ... (1 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/vcc', ... (4 bytes))
Mar 20 02:42:06 raspberrypi mosquitto[531]: Received PUBLISH from GarzsKapu (d0, q0, r1, m0, 'ESPURNA_21C176/status', ... (1 bytes))
....
Mar 20 02:42:29 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:42:29 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu
....
Mar 20 02:42:50 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:42:50 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu
....
Mar 20 02:43:11 raspberrypi mosquitto[531]: Received PINGREQ from GarzsKapu
Mar 20 02:43:11 raspberrypi mosquitto[531]: Sending PINGRESP to GarzsKapu

richlock1, 2018-03-25:
I made wireshark capture.
[electroDragonMqttDisconnet.zip](https://github.com/xoseperez/espurna/files/1845724/electroDragonMqttDisconnet.zip)

It seems there is a lot of re-transmission when the problem happens and both client and server takes much time inside this cycle. 

Good idea to increase ping timeout, less chance to get into this re-transmission situation, but I think the problem still be there. 
I will try.

xoseperez, 2018-04-04:
Any update on this?

xoseperez, 2018-04-12:
@richlock1 Can you try with latest 1.12.5? If the connection is stable and the keepalive interval is right I think we might close this issue.

richlock1, 2018-04-19:
I will try in the next couple of weeks, but I think it will be stable if you set big enough keepalive interval.

richlock1, 2018-05-29:
I have tried with 1.12.6 on electrodragon relay board. No problem at all. I used 300 secs keepalive interval. Thanks.

arihantdaga, 2018-06-06:
@xoseperez But setting keepAlive to 300 is also quiet unsafe, because the broker will not know if the device goes offline for atleast 150Secods i think.. right ? 
And it will show the device reachable even if it is actually not....

richlock1, 2018-06-06:
As I saw in the logs this setting affects publish timeout, which is 300 secs in our example, where Espurna reports its sensor values. The pingreq MQTT messages times are around 3.5 mins. It may be calculated some other way.