xoseperez/espurna #718: Lingan SWA1 Button Not Working

DJBenson, 2018-03-23:
I've got a generic plug from Amazon UK which turned out to be a Lingan SWA1. I've flashed the latest build of Espurna to it but the button does not work. The switch is otherwise fully functional.

    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

gn0st1c, 2018-03-24:
i'll try this with my SWA1 tonight.

gn0st1c, 2018-03-24:
@DJBenson : with your button1 definition, it works for me too. i've already have a PR fixing this.

btw, from now on, i'll also put the image of version information of the pcb (if available). this may help tracking issues faster/better.

e.g.
https://github.com/xoseperez/espurna/wiki/Hardware-Tonbux-XS-SSA06

DJBenson, 2018-03-24:
Glad it worked for you too.

xoseperez, 2018-03-24:
The PR looks good. Just a small glitch to be fixed and it will be merged.

DJBenson, 2018-04-16:
This plug is labelled as having power monitor capabilities, and there seems to be support for it in the OEM firmware - does anybody have any info on what is required (if anything) to get the power monitor to work with espurna?

xoseperez, 2018-04-16:
What power monitor chip does it have? I can't find any info on the Internet...

DJBenson, 2018-04-16:
That was the question really, there is a thread on a well-known UK bargain site which mentions that the official app has power monitoring built in but I can't find any evidence of their being power monitoring outside of that thread. I'm inclined to say it doesn't support it but then I'm intrigued how the WeConn app is reporting energy usage.

xoseperez, 2018-04-16:
No evidence. Maybe @gn0st1c can tell us if there is any suspicious chip inside...

gn0st1c, 2018-04-16:
the swa1 device i have, has two separate pcbs inside.

one with esp and other is power+relay. they are connected to each other with only 3 wires. (vcc, gnd, relay toggle). power side has only one IC, thx208. (switching power supply ic)

so at least for the one i got, there is no power monitor.

p.s.: due to some health issues, i should be away from light sources (computers etc) for a while. once i'm well again, i'll check the other swa1 clones i've to see if they are any different.

xoseperez, 2018-04-17:
Thanks @gn0st1c. Take care!

Closing, since this was merged with #721 into 1.12.5.

closed 2018-04-17 by xoseperez