xoseperez/espurna #719: The settings, even when never initialized, have priority over build #defines

PieBru, 2018-03-24:
Hi, this issue may confuse some Espurna newbies like me and may be blocking when the firmware is built without the WEB UI.

Example: ESPURNA_CORE built with MQTT_SUPPORT=1, MQTT_ENABLED=1 and MQTT_USE_JSON=1, used on a freshly erased Wemos D1 mini, does not send MQTT messages until the MQTT settings in WEB UI are manually updated and saved.
Also, the above settings initially appear OFF in the WEB UI, not reflecting what defined in the source code headers.

I've not digged in the code, but apparently a solution could be to initialize the settings structure with #defined values before reading it from the storage, but only if it has been saved at least once (valid signature/crc), thus not keeping hardware factory values as a valid Espurna config.

Just my 2 cents.
P.

xoseperez, 2018-03-24:
ESPURNA_CORE does not have Web UI by default. I guess you also enabled support for that. And then you need to specify an MQTT broker unless you have MQTT_AUTOCONNECT enabled and a discoverable (avahi / zeroconf/ bonjour) broker in your network.

The settings work just like you say. Check the getSetting method, it retrieves a value from the EEPROM and defaults to the "#define"d one.

PieBru, 2018-03-24:
On a virgin wemos d1 mini, getSetting() should not find any config, but the #defined values are ignored.
My custom.h contains only the wifi credentials, so after flashing the wifi connection works without WEB UI manuality.
When loading the WEB UI after flashing a virgin wemos, despite the #defines=1, the "Enable MQTT" and the "Use JSON payload" buttons remains OFF until manually set to ON and the config is saved.

And yes, I enabled WEB UI, this is my core_version.h
```
#undef WEB_SUPPORT
#define WEB_SUPPORT             1

#undef USE_PASSWORD
#define USE_PASSWORD            0

#undef MQTT_SUPPORT
#define MQTT_SUPPORT            1
#undef MQTT_ENABLED
#define MQTT_ENABLED            1
#undef MQTT_USE_JSON
#define MQTT_USE_JSON           1

#undef MQTT_SERVER
#define MQTT_SERVER             "192.168.1.10"
```

closed 2018-09-02 by xoseperez