xoseperez/espurna #720: NodeMCU/Espurna loosing track of time?

thenounoursk, 2018-03-24:
Hi Xosé, 

First of all, I should congratulate you for the work, your project is amazing! I've been using Espurna for a few weeks now and it's awesome. 

I've been having an issue recently that I thought might be interesting to share. I have two NodeMCU Lolin that I use with BME280 and MH-Z19B sensors. For some reasons, I can see there are periods where no readings are sent and the NodeMCU are not connected to my WiFi (as I can see through my router admin page). During those times, the NodeMCU still seem to be on though. I can see the led of sensor MH-Z19B still blinks, and the NodeMCU isn't in AP mode. Eventually after some time, the NodeMCU comes back on the network and starts sending sensor readings again. This doesn't seem to be a restart (see log below).

Precisely, in my NodeRED dashboard listening to mqtt messages, I see a temperature reading of 19.2 at 11:14, and then the next reading is 19.1 at 12:13. So I have a one hour gap. 

It was plugged to my computer with putty listening to serial. Here is the relevant section of the output log (full log attached): 

[476206] [MQTT] Sending /NodeMCUBedroom/temperature => 19.2 (PID 1211)
[476207] [MQTT] Sending /NodeMCUBedroom/pressure => 996.27 (PID 1212)
[476208] [MQTT] Sending /NodeMCUBedroom/humidity => 48 (PID 1213)
[476214] [MQTT] Sending /NodeMCUBedroom/co2 => 727 (PID 1214)
[476228] [MQTT] Publish ACK for PID 1211
[476232] [MQTT] Publish ACK for PID 1212
[476481] [MQTT] Publish ACK for PID 1213
[488460] [MQTT] TCP Disconnected
[476482] [MQTT] Publish ACK for PID 1214
[488460] [MQTT] Disconnected!
[488480] [WIFI] Connecting to BSSID: FF:3F:20:05:00:00 CH: 00, RSSI:   0, SEC: OPEN, SSID: mywifissid
[489776] [WIFI] MODE STA -------------------------------------
[489777] [WIFI] SSID  mywifissid
[489778] [WIFI] IP    192.168.0.241
[489778] [WIFI] MAC   68:C6:3A:84:1C:1E
[489781] [WIFI] GW    192.168.0.1
[489784] [WIFI] DNS   192.168.0.1
[489787] [WIFI] MASK  255.255.255.0
[489790] [WIFI] HOST  NodeMCUBedroom
[489793] [WIFI] BSSID C0:FF:D4:10:18:50
[489796] [WIFI] CH    6
[489799] [WIFI] RSSI  -50
[489801] [WIFI] ----------------------------------------------
[489807] [MDNS] OK
[490809] [MQTT] MQTT brokers found: 0
[490812] [MQTT] Connecting to broker at mymqttbrokerip:mqttport
[490812] [MQTT] Connecting as user mymqttuser
[490812] [MQTT] Client ID: ESPURNA_841C1E
[490814] [MQTT] QoS: 1
[490816] [MQTT] Retain flag: 1
[490819] [MQTT] Keepalive time: 240s
[490822] [MQTT] Will topic: /NodeMCUBedroom/status
[490839] [MQTT] Connected!
[490840] [MQTT] Unsubscribing to # (PID 1)
[490841] [MQTT] Subscribing to /NodeMCUBedroom/relay/+/set (PID 2)
[490842] [MQTT] Subscribing to /NodeMCUBedroom/led/+/set (PID 3)
[490846] [MQTT] Subscribing to /NodeMCUBedroom/action/set (PID 4)
[490855] [MQTT] Sending /NodeMCUBedroom/app => ESPURNA (PID 5)
[490857] [MQTT] Sending /NodeMCUBedroom/version => 1.12.4 (PID 6)
[490863] [MQTT] Sending /NodeMCUBedroom/host => NodeMCUBedroom (PID 7)
[490869] [MQTT] Sending /NodeMCUBedroom/ip => 192.168.0.241 (PID 8)
[490875] [MQTT] Sending /NodeMCUBedroom/mac => 68:C6:3A:84:1C:1E (PID 9)
[490881] [MQTT] Sending /NodeMCUBedroom/rssi => -51 (PID 10)
[490887] [MQTT] Sending /NodeMCUBedroom/uptime => 27490 (PID 11)
[490893] [MQTT] Sending /NodeMCUBedroom/datetime => 2018-03-23 11:15:05 (PID 12)
[490899] [MQTT] Sending /NodeMCUBedroom/freeheap => 18808 (PID 13)
[490905] [MQTT] Sending /NodeMCUBedroom/relay/0 => 0 (PID 14)
[490911] [MQTT] Sending /NodeMCUBedroom/vcc => 3061 (PID 15)
[490916] [MQTT] Sending /NodeMCUBedroom/status => 1 (PID 16)
[490931] [MQTT] Subscribe ACK for PID 2
[490932] [MQTT] Subscribe ACK for PID 3
[490932] [MQTT] Subscribe ACK for PID 4
[491164] [MQTT] Publish ACK for PID 5
[491165] [MQTT] Publish ACK for PID 6
[491165] [MQTT] Publish ACK for PID 7
[491165] [MQTT] Publish ACK for PID 8
[491167] [MQTT] Publish ACK for PID 9
[491170] [MQTT] Publish ACK for PID 10
[491173] [MQTT] Publish ACK for PID 11
[491177] [MQTT] Publish ACK for PID 12
[491180] [MQTT] Publish ACK for PID 13
[491184] [MQTT] Publish ACK for PID 14
[491187] [MQTT] Publish ACK for PID 15
[491190] [MQTT] Publish ACK for PID 16
[491927] [NTP] Time: 2018-03-23 12:12:49
[536433] [MQTT] Sending /NodeMCUBedroom/temperature => 19.1 (PID 17)
[536435] [MQTT] Sending /NodeMCUBedroom/pressure => 996.05 (PID 18)
[536435] [MQTT] Sending /NodeMCUBedroom/humidity => 48 (PID 19)
[536441] [MQTT] Sending /NodeMCUBedroom/co2 => 704 (PID 20)

So I see a TCP Disconnected event, which is suspicious, but not sure how to interpret this... Also, this TCP Disconnected event happens quite often during the day and most of the time it connects back immediately and I have no gap in sensor readings. Any clue to investigate this further? I have to admit I'm a bit clueless as to how investigate this further. 

[oputty2.log](https://github.com/xoseperez/espurna/files/1844367/oputty2.log)

Hopefully this is enough details. Obviously I'm happy to provide more if needed. 

Again, thank you. 

Florent

xoseperez, 2018-03-24:
The TCP Disconnect event does not seem to be related to any other event... are you using lwIP 2.0 or 1.4?

thenounoursk, 2018-03-24:
I'm using IwIP v2 Lower Memory. 

Other available options I see are:
- v2 Higher Bandwidth
- v1.4 Higher Bandwidth
- v1.4 Compile from source

Looking at your page https://github.com/xoseperez/espurna/wiki/ArduinoIDE I see you actually explain v2 has some instability issues. Sorry I must have overlooked it. What would v1.4 Pre built correspond to in the above options? 

Just tried to compile IwIP v.14 Higher Bandwidth which built fine. However, v1.4 Compile from source failed to build, complaining about "make" not being on my path, which should be easy enough to do. 

Thanks,
Florent

thenounoursk, 2018-03-24:
Well I should open my eyes. I suppose Pre built means going with v1.4 Higher Bandwidth. Giving this one a try on my NodeMCUs now. 

Will update you on this. 

Thanks,
Florent

thenounoursk, 2018-03-24:
Looks like moving to IwIP v1.4 Higher Bandwidth didn't do the trick. Within minutes of flashing, the same behaviour occurred again. I'll plug to the computer again to grab a log and check nothing new is in there. Unless there is another IwIP version that can be used?

thenounoursk, 2018-03-24:
Yeah, exactly the same behavior happens with lwIP v1.4 Higher Bandwidth. My NodeMCU is currently blocked. Log is being written and I'll post it when it unblocks but so far it looks exactly like that the previous one. That is, it shows nothing :( 

How can this lwIP be debugged into?

thenounoursk, 2018-03-24:
Log attach here, it just woke up.

[oputty4.log](https://github.com/xoseperez/espurna/files/1844767/oputty4.log)
 
I don't know how much value this will be, but the NodeMCU was blocked like described, and I got pissed off waiting, so I decided to blow gently on the BME280 sensor attached, because why not? And seconds after, the NodeMCU woke up and started communicating again. I don't believe in coincidences like this (it had been blocked for 1.5h or so), but at the same time, I have no idea how to connect the dots here. Would a sudden change in sensor values (humidity or temperature in this case) have any impact on the whole process?

thenounoursk, 2018-03-24:
Well I tried to blow on my second NodeMCU who also happens to be frozen and this one doesn't want to wake up. So maybe this was a coincidence after all.

closed 2018-08-13 by stale[bot]