xoseperez/espurna #723: SONOFF_RFBRIDGE max number of switches

moddsson, 2018-03-24:
I have been playing around with the number of  #define DUMMY_RELAY_COUNT parameter to increase the number of RF slots that I have for switching.
I manage to get the number increased up to **12** but it is downhill from there...
For some reason if for instance I set the number to **16** the RF option disappears from the sidebar when logging into the web interface.
According the the great writeup on Espurna and the Sonoff RF Bridge, there should be enough memory for 100 codes but I don't seem to be able to get it further than 12.

mcspr, 2018-03-26:
Sounds like websocket part of rfbridge module does not send its configuration. Or it sends it but the payload is partial.

This maybe unrelated, but I remember low heap situation of rfbridge when http://<ip>/config payload gets cut-off at exactly 1467 of 2305 bytes. Maybe large ammount of bridges sometimes half-delivers.

@moddsson If possible, can you modify onmessage here to `console.log(evt)` and see if json is actually partial? https://github.com/xoseperez/espurna/blob/1b1e58f0649a4a02e3337127220c3782c20e796d/code/html/custom.js#L1310
You can download espurna source release or clone this repo, add your changes and just open `code/html/index.html` in the browser. Then, in developer tools run `connect("<espurna-device-ip>")` and watch the log. (Note that for this to to work you first need to open real http://<espurna-device-ip> and login)

mcspr, 2018-04-04:
Just tried this with 16 relays. **relay.ino** sends 1768 byte message, **rfbridge.ino** sends 1010 on single client.
Opening web on 3 devices + telnet and rfbridge message starts disappearing.