xoseperez/espurna #725: css minification

gn0st1c, 2018-03-25:
[crass](https://www.npmjs.com/package/gulp-crass) is a better optimizer
[CSS minification benchmark](https://goalsmashers.github.io/css-minification-benchmark/)

cleancss:
`
329285 Mar 24 22:58 index.html.gz.h
`
crass:
`
328053 Mar 24 23:21 index.html.gz.h
`

https://github.com/xoseperez/espurna/blob/4dde84dfafa9bbfc5c840bd3e04c4f77e11f4d4b/code/gulpfile.js#L120

gn0st1c, 2018-03-25:
FYI, there are newer versions of the following packages.
i've been using them for a while without any issues so far.

+ del@2.2.2
+ gulp-clean-css@3.9.3
+ gulp-csslint@1.0.1
+ gulp-uglify@1.5.4
+ gulp-gzip@1.4.2
+ gulp-inline@0.1.3
+ gulp-htmllint@0.0.14

lobradov, 2018-03-28:
Care to create a PR? I'd love to test it.

gn0st1c, 2018-03-28:
Xose may be using older versions or different optimizer for a reason.

so instead of a PR, here are my files;
[gulp.zip](https://github.com/xoseperez/espurna/files/1857032/gulp.zip)

xoseperez, 2018-03-30:
Will test it for sure, more than 1Kb!! It almost looks like we are back in the 80's :smile:

xoseperez, 2018-03-30:
The size reduction after gzipping it is 248bytes, still something is something. I have noticed it requires strict mode and that means node 6.X or more.

gn0st1c, 2018-03-31:
i'm reopening this, this is the latest sizes;

before css class name refactoring:
`-rw-r--r--   1 gnostic  staff  65005 Mar 31 22:27 index.html.gz
`

after css class name refactoring:
`
-rw-r--r--   1 gnostic  staff  63843 Mar 31 22:30 index.html.gz
`

gn0st1c, 2018-03-31:
i hope to reduce more when i'm done with dce of js files :)

closed 2018-05-02 by xoseperez